RobotLocomotion/drake #17072: SOS optimization; how to determine which constraints are empty?

aabuaish, 2022-04-29:
Hi there,

I am using SOS optimization to solve an adaptive control problem using the inverse Lyapunov method. I have been successful in obtaining the Lyapunov function and region of attraction level-set for some simple problems. Now, I am trying to determine the Lyapunov for a new system. I am getting the error "Constraint ### is empty.", where ### is a number that changes randomly. How do I debug which constraint is empty? My constraints look like the following:
    prog.AddSosConstraint( V-l1 )
    prog.AddSosConstraint( -((beta-h)*p1 + V-1) )
    prog.AddSosConstraint( -(l2+Vdot) + p2*(V-1))
p1 and p2 have the decision variables. V, l1, and l2, are functions for the indeterminants only.
I am following the iterative procedure in [1] to solve for the Lyapunov function region of attraction level-set.

[1] F. Meng, D. Wang, P. Yang, G. Xie and F. Guo, "Application of Sum-of-Squares Method in Estimation of Region of Attraction for Nonlinear Polynomial Systems," in IEEE Access, vol. 8, pp. 14234-14243, 2020, doi: 10.1109/ACCESS.2020.2966566.

jwnimmer-tri, 2022-04-29:
Thanks for your interest!

We use Drake's issue database for acute bugs and feature requests.

For questions from users, we use the `#drake` tag on Stack Overflow instead: https://stackoverflow.com/questions/tagged/drake.  Please re-post your question into that forum.

(If you like, you can reply here with the Stack Overflow link, in case anyone who finds this issue wants to see the follow-up.)

I'll close this for now.  If we discover a bug in Drake as part of the Q&A, we can re-open it.

closed 2022-04-29 by jwnimmer-tri

aabuaish, 2022-04-29:
Wops. Thanks for the prompt response. I posted the question on Stack Overflow.