RobotLocomotion/drake #17073: Remove linux-focal-(unprovisioned-)clang-cmake(...) builds

jwnimmer-tri, 2022-04-29:
The review of RobotLocomotion/drake-ci#150 is to difficult to reason about with these builds in play.

Since they serve almost no purpose, we should just simplify our life and remove them from Jenkins.

jwnimmer-tri, 2022-04-29:
\CC @mwoehlke-kitware FYI

mwoehlke-kitware, 2022-04-29:
Isn't this no longer an issue?

It's a little odd that we ask Bazel how to set CC and then pass that back to Bazel, but AFAICT it's working fine. These builds verify that the compiler passed to CMake gets used by Bazel.

jwnimmer-tri, 2022-04-29:
It's quite possible that the updated CI scripting is correct.  However, covering this in CI has almost no value, and only serves to waste developer time and AWS coal.  Best to YAGNI on it.

mwoehlke-kitware, 2022-05-02:
...You trust the ability to use clang with the CMake build to not bit-rot, then?

BetsyMcPhail, 2022-05-10:
The following jobs have been removed:

**Provisioned Continuous:**

INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-continuous-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-continuous-everything-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-continuous-everything-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-continuous-release

**Unprovisioned Continuous:**

None

**Provisioned Experimental:**

INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-everything-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-everything-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-gurobi-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-gurobi-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-mosek-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-mosek-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-release **<-- pre-merge**
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-snopt-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-snopt-release

**Unprovisioned Experimental:**

INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-experimental-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-experimental-everything-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-experimental-everything-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-experimental-gurobi-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-experimental-gurobi-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-experimental-mosek-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-experimental-mosek-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-experimental-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-experimental-snopt-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-experimental-snopt-release

**Provisioned  Nightly**

INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-nightly-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-nightly-everything-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-nightly-everything-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-nightly-release

**Unprovisioned Nightly:**

INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-nightly-debug INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-nightly-everything-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-nightly-everything-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-unprovisioned-clang-cmake-nightly-release

**Weekly:**

INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-weekly-gurobi-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-weekly-gurobi-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-weekly-mosek-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-weekly-mosek-release
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-weekly-snopt-debug
INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-weekly-snopt-release

closed 2022-05-10 by BetsyMcPhail

jwnimmer-tri, 2022-05-10:
> INFO:jenkins_jobs.builder:Removing obsolete jenkins job linux-focal-clang-cmake-experimental-release <-- pre-merge

FYI I've just now removed this from the branch protection rule for `master`, so it's no longer pre-merge.