RobotLocomotion/drake #17079: Exceptions raised in a pyconstraint cause SnoptSolver to run forever

RussTedrake, 2022-05-02:
Adding `raise TypeError('error!')` to the beginning of the com_constraint in my [littledog notebook ](https://github.com/RussTedrake/underactuated/blob/96048316c6ec60a0b8fa917f06f98b8789131b2d/examples/littledog.ipynb) causes Snopt to run forever.

Expected behavior: The exception is reported to the console on the first call to the function.

I have tried capturing this in a more minimal example, but adding the same to [`test_pycost_and_pyconstraint`](https://github.com/RobotLocomotion/drake/blob/b4b243707f1ecd1f23e043b6fe05afaca51d92e9/bindings/pydrake/solvers/test/mathematicalprogram_test.py#L971) works as expected.

I will try to find a better reproduction, but this was a jarring enough error that I want to capture it before moving on.  Unfortunately, because some trajectory optimizations do take a long time to run, this was quite hard to debug initially.  (The original problem was a small syntax error in the constraint implementation that raised a TypeError, but ran forever).

I am running this on a mac, but don't know yet if that matters.

jwnimmer-tri, 2022-05-02:
The problem might be because SNOPT uses a C ABI, which cannot deal with C++ exceptions.  (I would have expected an `abort()` in that case, though.)

Possibly this patch would help surface the fault:
```patch
--- a/solvers/snopt_solver.cc
+++ b/solvers/snopt_solver.cc
@@ -534,6 +534,7 @@ void EvaluateAllNonlinearCosts(
   }
 }
 
+extern "C"
 void snopt_userfun(int* Status, int* n, double x[], int* needF, int* neF,
                    double F[], int* needG, int* neG, double G[], char* cu,
                    int* lencu, int iu[], int* leniu, double ru[], int* lenru) {
```

In any case, we should probably not pass `snopt_userfun` directly to SNOPT, but instead pass a function that wraps a `try ... catch ...` around it, instead.

RussTedrake, 2022-05-02:
Thanks @jwnimmer-tri.  Will try.
Note: I've confirmed that this issue does not happen for me on Ubuntu focal.

RussTedrake, 2022-05-02:
Unfortunately, adding `extern "C"` to snopt_solver.cc did not resolve the issue on mac.

RussTedrake, 2022-05-02:
Yes.  I aim to make that smaller reproduction.

RussTedrake, 2022-05-11:
The following code causes Solve to run forever on my mac:
```
import numpy as np
from pydrake.all import MathematicalProgram, Solve, CommonSolverOption, SolverOptions
prog = MathematicalProgram()
x = prog.NewContinuousVariables(2, "x")
prog.AddConstraint(np.abs(x[0]) + np.abs(x[1]) >= 1)

filename = "/tmp/debug.txt"
solver_options = SolverOptions()
solver_options.SetOption(CommonSolverOption.kPrintFileName, filename)
result = Solve(prog, solver_options=solver_options)
```

interestingly, `debug.txt` is written and says snopt "finished successfully".  
[debug.txt](https://github.com/RobotLocomotion/drake/files/8673839/debug.txt)

RussTedrake, 2022-05-11:
Here is another one that makes SNOPT run forever.  The constraint returns a scalar instead of a vector.
```
from pydrake.all import MathematicalProgram, Solve

def constraint(x):
    return x[0]  # return a scalar, instead of a vector

prog = MathematicalProgram()
x = prog.NewContinuousVariables(1, "x")
prog.AddConstraint(constraint, [0], [1], x)
result = Solve(prog)
```

jwnimmer-tri, 2022-05-11:
On Ubuntu...

For the first one I see:
```
jwnimmer@call-cps:~/jwnimmer-tri/drake$ env PYTHONPATH=/home/jwnimmer/Downloads/v1.2.0/drake/lib/python3.8/site-packages python3 foo.py
Traceback (most recent call last):
  File "foo.py", line 10, in <module>
    result = Solve(prog, solver_options=solver_options)
RuntimeError: NaN is detected during Symbolic computation.
```

For the second one I see:
```
jwnimmer@call-cps:~/jwnimmer-tri/drake$ env PYTHONPATH=/home/jwnimmer/Downloads/v1.2.0/drake/lib/python3.8/site-packages python3 foo.pyTraceback (most recent call last):
  File "foo.py", line 9, in <module>
    result = Solve(prog)
RuntimeError: PyFunctionConstraint: Output must be of .ndim = 1 or 2 (vector) and .size = 1. Got .ndim = 0 and .size = 1 instead.
```

This re-confirms that nice errors show up in Ubuntu, but apparently we hang on macOS.

jwnimmer-tri, 2022-05-11:
On an Intel mac with Drake v1.2.0 binary release ...

For the first one I see:
```
macsim:site-packages jwnimmer$ python3 foo1.py
libc++abi: terminating with uncaught exception of type std::runtime_error: NaN is detected during Symbolic computation.
Abort trap: 6
```

For the second one I see:
```
macsim:site-packages jwnimmer$ python3 foo.py
libc++abi: terminating with uncaught exception of type pybind11::error_already_set: RuntimeError: PyFunctionConstraint: Output must be of .ndim = 1 or 2 (vector) and .size = 1. Got .ndim = 0 and .size = 1 instead.
Abort trap: 6
```

I wonder why you don't see those?  Are you running on the command-line like I am here?

Are you running on an Intel mac, or under Rosetta, or native ARM64?

RussTedrake, 2022-05-12:
native ARM64

RussTedrake, 2022-05-12:
and I've been running in a jupyter notebook (from inside VSCode).  Just tried from the console:
```
russt@TRI-FVFG13URQ6LT Downloads % python3
Python 3.9.12 (main, Mar 26 2022, 15:45:34)
[Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from pydrake.all import MathematicalProgram, Solve
>>>
>>> def constraint(x):
...     return x[0]  # return a scalar, instead of a vector
...
>>> prog = MathematicalProgram()
>>> x = prog.NewContinuousVariables(1, "x")
>>> prog.AddConstraint(constraint, [0], [1], x)
<pydrake.solvers.mathematicalprogram.Binding[Constraint] object at 0x102c4d530>
>>> result = Solve(prog)
zsh: trace trap  python3
```
so that returns immediately (with the trap)

my normal jupyter install is borked, so I'm afraid I can't test that til i'm back on a better internet connection

jwnimmer-tri, 2022-05-12:
Okay, the "run forever" might be a Jupyter-specific failure mode of C++ shutdown / crashing.

In any case, most likely the most important fault to try to address here is that the macOS EH ABI does not allow C++ exceptions to jump over C code, so any user-cost-function exceptions (whether thrown from Python or C++) will take down the entire program or Python interpreter.  (And sometimes, that locks up while crashing, inside Jupyer.)

Fixing that EH ABI would probably mean adding more glue code to `snopt_solver.cc` to add `try...catch` around every callback and turn faults there into `setjmp/longjmp`, a bit like we did for CSDP recently.

hongkai-dai, 2022-05-25:
@jwnimmer-tri this issue is assigned to me, but since you already looked into the issue, would you like to take it over? I am not familiar with setjmp/longjmp.

jwnimmer-tri, 2022-05-26:
Sure, sounds good.  I think we've proved that this is a macOS-specific ABI issue, not anything to do with SNOPT features, specific program formulations, etc.

jwnimmer-tri, 2022-05-26:
The shape of the fix will be to take this point in the code:

```c++
  // Actual solve.
  const char problem_name[] = "drake_problem";
  // clang-format off
  Snopt::snkera(Cold, problem_name, nF, nx, objective_constant, ObjRow,
```

and call `setjmp`, and then take this place in the code:

```c++
void snopt_userfun(int* Status, int* n, double x[], int* needF, int* neF,
                   double F[], int* needG, int* neG, double G[], char* cu,
                   int* lencu, int iu[], int* leniu, double ru[], int* lenru) {
```

wrap it in an internal try-catch, and in the catch do a `longjmp`.

Probably also try to stash the current exception message somewhere the user would see it.  Ideally we'd re-throw it from the snkera location, but at worst we could live with printing it out.

We should probably limit this dance to macOS only.  That would mean less coverage of this control flow, but also better UX on Ubuntu.  Probably worthwhile to limit to macOS, but we could change our mind on that as we go.

In the meantime, we could imagine a band-aid on macOS where we `try-catch`, log the error message, and `abort()`.  That might at least make the error message more prominent.

PR #17272 (2022-06-08): [solvers] Manually propagate exceptions within SNOPT
Without this, the macOS EH ABI cannot cross a C++/C boundary and would otherwise abort.

Closes #17079.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17272)
closed 2022-06-08 by sherm1