RobotLocomotion/drake #17080: Improve error message when quaternion state is set to all zeros

RussTedrake, 2022-05-02:
The following code
```
from pydrake.all import *

initial_state = [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
# instantiate diagram builder used to connect the different system components
builder = DiagramBuilder()

# define world multibody plant 
world_plant, scene_graph = AddMultibodyPlantSceneGraph(builder, time_step = 0.0)

#### Define quadrotor MultiBodyPlant using Quadrotor .urdf and Propeller class
quad_model= Parser(world_plant).AddModelFromFile(FindResourceOrThrow("drake/examples/quadrotor/quadrotor.urdf"))

#finalize the world plant
world_plant.Finalize()

# Set up a simulator to run this diagram
diagram = builder.Build()
simulator = Simulator(diagram)
sim_context = simulator.get_mutable_context()
sim_context.SetContinuousState(initial_state)

simulator.AdvanceTo(0.1)
```
fails because the state is set to zero, despite it having a quaternion floating base.

But the error message is
```
RuntimeError: Encountered singular articulated body hinge inertia for body node index 1. Please ensure that this body has non-zero inertia along all axes of motion.
```
which leads people to debug in all of the wrong places.

Note: I had mentioned this in https://github.com/RobotLocomotion/drake/issues/16992#issuecomment-1114311140, but have now found this to be the root cause and a minimal reproduction.

RussTedrake, 2022-05-02:
It turns out that this error can crop up in a much more insidious way, too.  If someone constructs a `MultibodyPlant` that has a floating base and passes it directly to our trajectory optimization codes (without explicitly adding unit quaternion constraints), then the optimizer can easily drive the solution into these singular states.  That is *much* harder to debug, and explains some of the issues that have been coming up in class projects.

Because this is an awful trap to lay for users, I will raise this to priority medium.

jwnimmer-tri, 2022-05-02:
FYI tangentially related to #14800.

sherm1, 2022-05-02:
@mitiguy do you want to add this to your list of make-Drake-error-messages-mean-something tasks?

mitiguy, 2022-05-02:
Yes -- please assign me.

mitiguy, 2022-07-08:
PR https://github.com/RobotLocomotion/drake/pull/17399 resolves issue https://github.com/RobotLocomotion/drake/issues/17080

closed 2022-07-08 by mitiguy