RobotLocomotion/drake #17087: [WIP][SAP] Parameterization of contact dissipation

amcastro-tri, 2022-05-03:
Unlike [TAMSI](https://arxiv.org/abs/1909.05700), which uses a Hunt & Crossley (H&C) model of dissipation, [SAP](https://arxiv.org/abs/2110.10107) uses a linear model of dissipation.
For point contact of stiffness `k` the compliant normal force fₙ is modeled as:
>  fₙ=(−k⋅(ϕ+τ⋅vₙ))₊

where (x)₊=max(0,x), ϕ is the signed distance, vₙ the normal velocity and we define the "dissipation time scale" τ, the topic of this PR. (For standard notation like ϕ, vₙ, etc. please refer to [[Castro et al. 2019]](https://arxiv.org/abs/1909.05700) and/or [[Castro et al. 2021]](https://arxiv.org/abs/2110.10107)).
Similarly for hydroelastic contact we model dissipation as:
>  pₙ=(pₑ+τ⋅ṗ)₊

where pₑ is the "elastic contribution" result of the balance of pressure fields and ṗ is the rate of change of the pressure field at a given location on the surface, which is shown to be [[Masterjohn et al., 2022]](https://arxiv.org/abs/2110.04157)
>  ṗ=g⋅vₙ,

with `g` the "effective pressure gradient", see [[Masterjohn et al., 2022]](https://arxiv.org/abs/2110.04157).

## Analogy to Harmonic Oscillator
As soon as a modeler sees those equations, their brain should automatically scream "harmonic oscillator"!. If not please read [wiki summary](https://en.wikipedia.org/wiki/Harmonic_oscillator) or refer to classic physics books such as [[Halliday & Resnick]](https://www.amazon.com/Fundamentals-Physics-Binder-Ready-Version/dp/1118230647) or [[Landau & Lifshitz]](https://www.amazon.com/Mechanics-Course-Theoretical-Physics-Landau/dp/0750628960/ref=sr_1_1?keywords=landau+and+lifshitz&qid=1651587950&s=books&sprefix=landau+an%2Cstripbooks%2C52&sr=1-1).

The analogy is evident if we consider a particle of mass `m` **in contact** (and thus fₙ > 0) with the ground. Then the equations of motion for that particle are:
> mv̇ₙ = −k⋅(ϕ+τ⋅vₙ) = −k⋅ϕ-c⋅vₙ

which are the harmonic oscillator equations for a mass `m` moving along the normal coordinate (in this case parameterized by signed distance ϕ) with dissipation constant `c = τ⋅k`. Clearly those equations are not longer valid when we break contact, but the analogy is very helpful to think about the contact phase.

## Time Scale Parameterization

Some very important aspects to point out:
  1. This model is NOT specific to SAP.  It is a physical modeling choice (with pros and cons that I believe do not belong to this issue). For instance, It'd be very simple to implement this model in TAMSI. The fact that SAP does not support H&C, does not mean that linear dissipation is specific to SAP.
  2. The parameterization in terms of the "dissipation time scale" is very useful in that it allows to parameterize linear dissipation for both point contact and hydroelastic contact.
  3. Zero time scale, τ=0, corresponds to zero dissipation (c=0 for the harmonic osscillator). Do not think of τ in terms of "how fast dissipation acts" or you'll get the opposite intuition. 

## SDF/URDF Parsing
To allow users to

jwnimmer-tri, 2022-05-03:
In general, for parser diagnostics we've been using the following rules:

(1) When the input file is **not** consistent with the allowed syntax (e.g., misspelled element names, mismatched tags, etc.), then that is an Error condition.

(2) When the input file **is** consistent with the allowed syntax, but Drake is not going to obey a command in the input file  -- either because it refers to an unimplemented feature (e.g., "gearbox"), or the value is irrelevant in the current configuration (e.g., "mu_dynamic"), etc. -- then that is a Warning condition only.  (The user can opt-in to promote Warnings to Errors, if they wish.)

We prefer to use Warning in those cases because it gives the user important feedback they need, but does not get in the way of them working on other problems of the plant or simulation before cycling back to deal with the parsing complaints.  We leave it up to them to judge whether the diagnostic is immediately relevant.  If we Error'd on those things, we'd be forcing them to deal with reports that might not matter to them yet.

(I should probably spell this out in our docs somewhere.  Maybe the Parser overview @rpoyner-tri?)

amcastro-tri, 2022-07-12:
Regarding naming of the dissipation timescale. This model I propose here is nothing but a [Kelving-Voigt model of dissipation](https://en.wikipedia.org/wiki/Kelvin%E2%80%93Voigt_material). The analogy is even more obvious for hydroelstic, which is written in terms of pressure (normal stress). 

If you follow [this link](https://en.wikipedia.org/wiki/Retardation_time) there are a couple more options that I actually like a lot:
 1. Retardation time
 2. Relaxation time

I am particularly fond of relaxation time, since I always explain the "dissipation timescale" to people in the same way: "apply stress to your material and then let it go. The time it takes for the material to *relax* to the underformed configuration is a good proxy of this timescale". Therefore "relaxation time" or "relaxation timescale" seems more appropriate? it is nice also to find this is a term of art for an already existing model, I didn't come up with anything new here.

@sherm1, @mitiguy, WDYGT?

sherm1, 2022-07-12:
I ❤ `relaxation_time` !!

xuchenhan-tri, 2022-07-12:
"relaxation time" sounds good to me. It agrees with that used in my go-to reference _Computational Inelasticity by Simo and Hughes_ (Sec. 1.7.1)

mitiguy, 2022-07-12:
The term "resonates" with me :)

jwnimmer-tri, 2022-07-19:
I'm not sure what work remains here, but it any case it sounds like MbP work that @amcastro-tri will handle, not Parser work where @rpoyner-tri is accountable for anything.  Updating issue labels to match.

amcastro-tri, 2022-07-19:
Actually we can close this! resolved in #17554. Thanks

closed 2022-07-19 by amcastro-tri