RobotLocomotion/drake #17112: Add hardware_station_simulation capability

jwnimmer-tri, 2022-05-05:
Details to follow.

(Cross-reference for Anzu issue 8760.)

jwnimmer-tri, 2022-05-25:
One question here is how to provide a demo of this feature, with an example binary.  If we can figure it out, it might be nice to replace `kuka_simulation` with this program, if we can?  @sammy-tri has some ideas.

@calderpg-tri also suggests that we offer a 2-arm config file in the demo as well.

mpetersen94, 2022-07-20:
Just found this issue as I was trying to implement a similar setup for my lab's research.  Would love even a suggestion of good ways to organize the config file so that the simulation/communication layer is adjustable for different numbers/types of robots!

jwnimmer-tri, 2022-07-20:
We are getting very close to publishing this software (active work for the past ~month).  In the meantime, here's a sample config files for what we use:  https://gist.github.com/jwnimmer-tri/b087cc01f41296d637fe33c032762298

The glue code that supports `model_drivers` and `lcm_buses` will be some of the first to arrive.

mpetersen94, 2022-07-20:
The config file is already super helpful! Does the directive file have all the models to load into the simulator (i.e. arms, grippers, welded objects and floating objects)? I'm also assuming the model drivers section is telling what control stack (that would run on the real robot) to spoof for the simulation? Feel free not to answer if any of this is proprietary.

jwnimmer-tri, 2022-07-20:
Yes, exactly right on both counts.

For loading models into the plant (environment, robots, manipulands), we tend to use cascading directives files -- the typical top-level one for a station includes another one (e.g., our standard iiwa or panda) which includes another one (how the gripper is typically welded), etc. That allows for some amount of configuration re-use.

The model_drivers basically wire up the systems such as https://github.com/RobotLocomotion/drake/tree/master/manipulation/kuka_iiwa along with inverse dynamics to respond to lcm commands and report status, just like the real robot.

jwnimmer-tri, 2022-07-21:
See #17615 for a demo program underway.

mpetersen94, 2022-07-27:
Now that #17615 has merged, what is the missing glue that is needed to implement a similar hardware simulator on the python side?

jwnimmer-tri, 2022-07-27:
I am not entirely sure, but I do plan to work on it (and have a Python twin to that demo as an example).

I'm guessing that if you wrote your own `scenario.{h,cc}` along with `LoadScenario` in C++ and bound that into Python, then you could call all of the `Apply...` functions one by one that are already bound in pydrake.

I think it should also be possible to have a python-only `scenario.py` with no native code at all, but that probably requires a little more infrastructure that isn't provided yet.

jwnimmer-tri, 2022-08-09:
@zachfang I updated the original post atop this issue with a checklist of remaining work.  LMK if you think anything is missing.

zachfang, 2022-08-09:
One thing I can think of is the extra MultibodyPlant functions. And they will need some more thorough unit tests as well.

jwnimmer-tri, 2022-10-10:
#18066 contains my first draft of the Python version.  It's not feature-complete yet, but I think it's worth landing a first version and then adding the final, more-difficult features.

jwnimmer-tri, 2023-01-19:
Okay, the code is done now (I think).  What remains is to make a better introduction / demo / tutorial / etc, whatever form that takes. At minimum, it would be an expanded README.

jwnimmer-tri, 2023-02-24:
I looked at this some today.  I think the best answer is to:
- Update the scenario file to be self-contained (fold `example_iiwa_wsg.dmd.yaml` back into the main scenario file).
- Add 1-2 more scenarios (pendulum?).
- Explain to users how to copy 2 files into their own pip env and play with the demo.

PR #18908 (2023-03-06): [examples] Finish documenting the hardware_sim example
Fold the sample data into a single file. Nested model directives are interesting, but not the purpose of this example.

Switch the sample files to MIT-0 license, so that people who copy it don't need to preserve our BSD-3-Clause license text.

---

Closes #17112.

You can see a preview of the README here:

https://github.com/jwnimmer-tri/drake/blob/hardware-sim-example/examples/hardware_sim/README.md#standalone-robot-simulations

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18908)