RobotLocomotion/drake #17118: show_model does not work for composed models

JoseBarreiros-TRI, 2022-05-05:
Show_model is currently not supporting composed models (e.g. those that use the `<include>` tag to include models defined in a different file or have two models nested in the world). This is a powerful feature of the [SDFormat](http://sdformat.org/tutorials?tut=composition).

When testing it with a composed model, the following error is trow: `RuntimeError: File must have a single <model> element.`

sherm1, 2022-05-06:
Assigned to Eric for disposition.

EricCousineau-TRI, 2022-05-09:
I've submitted #17129 - most importantly, it does support composite **models**.
What it does not support is *worlds* with composite models.

@JoseBarreiros-TRI My PR permits it to use worlds; however, I think it is *vital* that your composition for your model to be done *as a model*, not as a world. In that way, you can instantiate your model in different contexts.
(In SDFormat meetings, we've discussed being able to nest worlds, but I'm going to strongly assert it's a waste of time ;)

Does this make sense?

JoseBarreiros-TRI, 2022-05-09:
Thanks @EricCousineau-TRI. This should address my request and noted on your recommendation about model composition. I agree that the composition should be done as a model, not as a world.

PR #17129 (2022-05-09): show_model: Test composite model; also enable usage of world files
Resolves #17118

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17129)
closed 2022-05-09 by sherm1