RobotLocomotion/drake #17124: Changes to CI schedule frequency

jwnimmer-tri, 2022-05-07:
- `linux-(focal|gcc)-clang-bazel-nightly-everything-coverage` switch to Weekly instead of Nightly.
  - This is unlikely to fail individually, and is expensive.
- `linux-focal-unprovisioned-clang-bazel-continuous-(debug|release|everything-debug|everything-release)` remove.
  - We already have it in Nightly, and we don't need fast feedback here.
- `linux-focal-unprovisioned-gcc-bazel-continuous-(debug|everything-debug|everything-release)` remove.
  - We already have it in Nightly, and we don't need fast feedback here.
  - Note that `linux-focal-unprovisioned-gcc-bazel-continuous-release` stays as Continuous.
- `linux-focal-(clang|gcc)-cmake-continuous-(debug|everything-debug|everything-release)` remove.
  - We already have it in Nightly, and we don't need fast feedback here.
  - Note that `linux-focal-(clang|gcc)-cmake-continuous-release` stays as Continuous.
- `linux-focal-clang-bazel-continuous-everything-(debug|release)` remove.
  - We already have it in Nightly, and we don't need fast feedback here.
- `linux-focal-gcc-bazel-continuous-everything-debug` remove.
  - We already have it in Nightly, and we don't need fast feedback here.
- `mac-monterey-unprovisioned-clang-bazel-continuous-snopt-mosek-packaging` remove.
  - We already have it Weekly, and we don't use these tarballs for anything.

jwnimmer-tri, 2022-05-07:
If my math is correct, that will leave only these jobs in Continuous:
- `linux-focal-clang-bazel-continuous-debug`
- `linux-focal-clang-bazel-continuous-everything-address-sanitizer`
- `linux-focal-clang-bazel-continuous-everything-leak-sanitizer`
- `linux-focal-clang-bazel-continuous-release`
- `linux-focal-gcc-bazel-continuous-debug`
- `linux-focal-gcc-bazel-continuous-everything-release`
- `linux-focal-gcc-bazel-continuous-release`
- `linux-focal-gcc-cmake-continuous-release`
- `linux-focal-unprovisioned-gcc-bazel-continuous-documentation`
- `linux-focal-unprovisioned-gcc-bazel-continuous-mirror-to-s3`
- `linux-focal-unprovisioned-gcc-bazel-continuous-release`
- `linux-focal-unprovisioned-gcc-bazel-continuous-snopt-mosek-packaging`
- `mac-big-sur-clang-bazel-continuous-release`
- `mac-big-sur-clang-cmake-continuous-release`
- `mac-big-sur-unprovisioned-clang-bazel-continuous-snopt-mosek-packaging`
- `mac-monterey-clang-cmake-continuous-release`

BetsyMcPhail, 2022-05-10:
Note that linux-focal-clang-cmake jobs were all removed in #17073

BetsyMcPhail, 2022-05-10:
The Continuous list in Jenkins now matches with what is listed above, with the exception of `mac-monterey-clang-bazel-continuous-release`  which has _not_ been removed yet, see my previous comment.

jwnimmer-tri, 2022-05-10:
> mac-monterey-clang-cmake-nightly-release doesn't exist. Should this be mac-big-sur-clang-cmake-continuous-release?

I'm not sure what I was thinking, but yes this sounds better.  I think my goal was for only one mac cmake build to run in Continuous.  Seems like big-sur is the right choice for that.

BetsyMcPhail, 2022-05-10:
Changes to drake-jenkins-job are at [f779025](https://github.com/RobotLocomotion/drake-jenkins-jobs/commit/f77902524b94ab44a7aa7bb515a0da9a8fe70c38)

closed 2022-05-10 by jwnimmer-tri