RobotLocomotion/drake #17130: Add Polynomial::Expand() function

hongkai-dai, 2022-05-09:
We have `Expression::Expand()`. I think it is natural to add `symbolic::Polynomial::Expand()`
```cc
symbolic::Polynomial Polynomial::Expand();
```
which expands all the coefficients for every monomial, and returns the expanded polynomial.

One question I have is about monomial with 0 coefficient. From the user's perspective, is it more natural to keep or remove the monomial with 0 coefficient after expansion? I am leaning towards removing the zero-monomial as it is redundant, and `Expression::Expand()` function does a lot of work to remove the redundancy. But I would like to hear the user's opinion.

@jeremyma-tri what do you think?

jwnimmer-tri, 2022-05-09:
My impression is that `symbolic::Polynomial` maintains an invariant that no coefficient can ever be zero.  There is a lot of code to already enforce that (though there are a few glaring errors where we fail to preserve it, e.g., `Polynomial::EqualToAfterExpansion`).  I think we should strip out zero coefficients.

hongkai-dai, 2022-05-09:
>  (though there are a few glaring errors where we fail to preserve it, e.g., Polynomial::EqualToAfterExpansion).

Yeah I also notice that EqualToAfterExpansion doesn't strip zero. I will fix it.

closed 2022-05-12 by jwnimmer-tri