RobotLocomotion/drake #17131: Toppra with time-varying limits

sadraddini, 2022-05-09:
The current implementations of [AddFrameAccelerationLimit](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_toppra.html#a4439ae2c06e737634078434d897cceaf) and [AddFrameTranslationalSpeedLimit](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_toppra.html#a5cee3da4435b14492b675df2a473f611) add Cartesian acceleration/velocity limits on a specific frame of the robot throughout the whole trajectory. These constraints are often not related to the physical capabilities of the robot (even though Franka [does have Cartesian limits](https://frankaemika.github.io/docs/control_parameters.html) - apparently for user safety) but are specific to user application. What we wish is to make Toppra accept such constraints in a non-uniform way throughout the trajectory. For instance, we need the robot end effector to slow down in a portion of the trajectory while not having such a constraint on the rest. Of course, breaking the problem into segments and solving Toppra independently is not the ideal solution as the optimal solution requires the whole trajectory to run through Toppra once.

Per discussions with @jwnimmer-tri, I think the first step is to agree on the following forms of header functions. I think of overloading the current ones in this way: 
for velocity:
```
  /**
   * Adds a limit on the magnitude of the translational velocity of the given
   * frame, measured and expressed  in the world frame, as a function of path
   * parameter.
   * @param constraint_frame The frame to limit the translational speed of.
   * @param upper_limit The upper translational speed limit trajectory for
   *                    constraint_frame.
   * @return The bounding box constraint that will enforce the frame
   *         translational speed limit during the backward pass.
   */
  // Note: the constraints are only added between upper_limit.start_time() and
  //       upper_limit.end_time(). The rest of the time is not constrained.
  //       A warning is displayed if upper_limit.start_time() <
  //       path.start_time() or upper_limit.end_time() > path.end_time().
  Binding<BoundingBoxConstraint> AddFrameTranslationalSpeedLimit(
      const Frame<double>& constraint_frame,
      const Trajectory<double>& upper_limit);
```
for acceleration:
```
  /**
   * A version of accleration limit that uses a trajectory for the upper and
   * lower limits.
   * @param constraint_frame The frame to limit the acceleration of.
   * @param lower_limit The lower acceleration limit trajectory for
   * constraint_frame.
   * @param upper_limit The upper acceleration limit trajectory for
   * constraint_frame.
   * @param discretization The discretization scheme to use for this linear
   *                       constraint. See ToppraDiscretization for details.
   * @return A pair containing the linear constraints that will enforce the
   * frame acceleration limit on the backward pass and forward pass
   * respectively.
   */
  // Note: the constraints are only added between upper_limit.start_time() and
  //       upper_limit.end_time(). The rest of the time is not constrained.
  //       A warning is displayed if upper_limit.start_time() <
  //       path.start_time() or upper_limit.end_time() > path.end_time().
  std::pair<Binding<LinearConstraint>, Binding<LinearConstraint>>
  AddFrameAccelerationLimit(const Frame<double>& constraint_frame,
                            const Trajectory<double>& lower_limit,
                            const Trajectory<double>& upper_limit,
                            ToppraDiscretization discretization =
                                ToppraDiscretization::kInterpolation);
```

jwnimmer-tri, 2022-05-09:
\CC @mpetersen94 and @hongkai-dai please let us know if you have any comments on the proposed API, above.  We plan to implement this soon.

---

> ...the note comment between the doxygen comment...

Right, I'd say that the first part of the note is an important reminder for users of the API, so we should put it into the Doxygen directly:

```c++
/**
 * @param ...etc...
 *
 * @note The constraints are only added between upper_limit.start_time() and
 *  upper_limit.end_time(). The rest of the time is not constrained. */
```

More interesting is the comment about the time boundaries:

```c++
//       A warning is displayed if upper_limit.start_time() <
//       path.start_time() or upper_limit.end_time() > path.end_time().
```

In general, I don't think code should rely on `log()->warn(...)` for identifying logic errors.  Some users will not see any warnings (due to their runtime environment), and furthermore there's no programmatic way to catch and respond to those diagnostics in code.  We have to rely on user eyeballs staring at the screen.

I'd like to either promote this condition to a hard error (throwing an exception), or else handle it silently (by ignoring the trajectory outside of the plan's times).  I'm happy to take input on which one is better.

jwnimmer-tri, 2022-05-09:
It seems to me like the proposed `AddFrameTranslationalSpeedLimit` should offer both `lower_limit` and `upper_limit`, to remain consistent with all of the other API on Toppra.  The user can pass `-inf` if they want it to be unlimited.

hongkai-dai, 2022-05-09:
For the velocity part it says "Adds a limit on the magnitude of the translational velocity of the given
   * frame, measured and expressed  in the world frame, as a function of path
   * parameter." I am not sure what "magnitude" here means, is it the the l2-norm of the ℝ³ vector of translational velocity, namely this magnitude is a non-negative scalar? Or is it the absolute value of each x/y/z entry of the translation velocity, namely this magnitude is also an ℝ³ vector with non-negative entries?

sadraddini, 2022-05-09:
> I am not sure what "magnitude" here means, is it the the l2-norm of the ℝ³ vector of translational velocity, namely this magnitude is a non-negative scalar

Yes, it means the norm. The idea is to constrain the speed (norm of velocity vector)

hongkai-dai, 2022-05-10:
Thanks Sadra, I think the API looks good. Would be great if @mpetersen94 could take a look as well.

mpetersen94, 2022-05-10:
Overall I think the API looks good with the reformatting of documentation Jeremy suggested.  I also agree with Jeremy that
//       A warning is displayed if upper_limit.start_time() <
//       path.start_time() or upper_limit.end_time() > path.end_time().
should either be ignored or produce a hard error. I lean towards ignored but that might make debugging hard if the limit trajectory doesn't overlap with the path trajectory.

jwnimmer-tri, 2022-10-17:
> I'd like to either promote this condition to a hard error (throwing an exception), or else handle it silently (by ignoring the trajectory outside of the plan's times).

If we decide to make it a hard error, and then later find that it's too finicky, it's easy to switch the contract to "ignore".  If start with "ignore" as the contract and then later decide we want an error, it will be a slightly more difficult transition.

On the other hand, it doesn't seem like a problem to have "ignore" as the behavior.

Maybe this: in case the two trajectories are completely disjoint, then that's a hard error (because the constraint is a no-op -- unlikely what the user wanted).  But if there's any overlap at all, then we just ignore the new limits outside of path.

PR #18738 (2023-02-13): [multibody] Adding time-varying constraints to two Toppra APIs
This allows setting the cartesian frame speed and acceleration limits with trajectories.

Resolves #17131 

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18738)
closed 2023-02-13 by SeanCurtis-TRI