RobotLocomotion/drake #17132: Tutorial on collisions / proximity

jwnimmer-tri, 2022-05-09:
The very basics of collision handling are explained in our MbP tutorials, but there is a lot more to learn beyond the basics.  We should probably create a tutorial specifically on collisions and proximity topics.

- Collision filtering concepts, APIs, parsing, etc.
- How to make signed distance queries
- Geometry ID vs Body ID etc.
- Non-MbP geometries
- Displaying collision geometries
- Displaying collision results

CWEzio, 2022-05-10:
This is what I am looking for! There seem to be not much examples or tutorials regarding the usage of geometry. The document for [geometric representation](https://drake.mit.edu/doxygen_cxx/group__geometry.html) is incomplete and seems haven't been updated for a while.

SeanCurtis-TRI, 2022-06-07:
Quick question: 

> Non-MbP geometries

What's the provenance of this? Currently, there is very little to be said of the subject (contact with any non-MbP geometry leads to failure). Was there a particular circumstance where this came up? Or does it simply need to be explicitly addressed?

jwnimmer-tri, 2022-06-07:
There are several examples of calls to `SceneGraph::RegisterSource` besides MbP, e.g., `examples/acrobot/acrobot_geometry.cc`.

jwnimmer-tri, 2022-06-07:
More likely, what I was thinking about at the time was the (forthcoming) voxelized collision checker, which does not use SceneGraph per se.  However, since it's not actually shipped yet, we can't add it to the tutorial anyway.

SeanCurtis-TRI, 2022-06-07:
Very helpful clarification. I was interpreting it through a different lens. The topic of "using `SceneGraph` without `MbP`" is definitely a meaningful one.

jwnimmer-tri, 2026-08-21:
FYI @SeanCurtis-TRI the new "inactive" #24614 feature might be another relevant topic for here.