RobotLocomotion/drake #17137: Better testing of limit constraints

amcastro-tri, 2022-05-09:
As thoroughly discussed in #17083, the unit testing of the `CompliantContactManager::AddLimitConstraints()` can be further improved.
In particular @xuchenhan-tri pointed out that:
 1. Changing the value of `kPositiveDeltaQ` can break the unit test. How this value is selected should at least be documented or better yet, make the unit test not sensitive to arbitrary values of this testing parameter.
 2. Ideally we would like the testing conditions enumerated by the enum `InitializePositionAt` to be mutually exclusive. 

The purpose of this issue is to lay down a better testing strategy and to track progress towards landing this new strategy into Drake master.

amcastro-tri, 2022-05-12:
@xuchenhan-tri wrote [this patch](https://github.com/xuchenhan-tri/drake/commit/eb8bf5ac353e586387ede71b277f945bf983bebe) that addresses his concerns.

Those concerns inlude:
  1. The fact that `AboveLowerLimitThoughPredictionBelow` leads to only one constraint depends on how large the velocity is. This should at least be documented in the test.
  2. Whether a predicted value is beyond a given limit, will depend also on the `kLimitWindowFactor`. Therefore the test should sanity check that value and document any assumption it might be making in that regard.
  1. In `compliant_contact_manager_test.cc` line 1450, `lower_limit_expected` and `upper_limit_expected` are exclusive (at least with the velocities selected), so we'll never have two constraints.

amcastro-tri, 2022-05-12:
with `(wl, wu)`, the predicted window and
with `(ql, qu)` the actual limits.

Then I have four conditions:
1. wl < ql  (only lower)
2. wu > qu  (only upper)
3. ql < wl < wu < qu (predicted window completely inside limits. No limits)
4. wl < ql < qu < wu (limits inside predicted window. Both limits)

xuchenhan-tri, 2024-03-26:
I'm going to close this issue as unplanned. Feel free to reopen if there's interest to carry my patch cited above forward.

closed 2024-03-26 by xuchenhan-tri