RobotLocomotion/drake #17139: UB in picosat

jwnimmer-tri, 2022-05-09:
During the Clang 12 upgrade (#16937) we found that picosat contains undefined behavior in it's C code.

Refer to https://github.com/dreal-deps/picosat/pull/1 for a proposed fix.

For the moment, we've worked around it (by disabling the sanitizer).  This issue is to track the correct fix (to the code itself).

jwnimmer-tri, 2022-06-30:
Running `bazel test --config=clang --compilation_mode=dbg --config=ubsan //solvers/...` using this patch to pull in that PR:

```patch
diff --git a/tools/dynamic_analysis/bazel.rc b/tools/dynamic_analysis/bazel.rc
index d8ddacd332..dc1a8b5485 100644
--- a/tools/dynamic_analysis/bazel.rc
+++ b/tools/dynamic_analysis/bazel.rc
@@ -215,10 +215,6 @@ build:ubsan_everything --test_lang_filters=-sh,-py
 # See https://developer.apple.com/documentation/code_diagnostics/undefined_behavior_sanitizer
 build:ubsan_everything --test_timeout=120,600,1800,7200
 
-# TODO(#17139) Fix picosat to be UBSan-correct.
-build:ubsan --per_file_copt=external/picosat/picosat.c@-fno-sanitize=pointer-overflow
-build:ubsan_everything --per_file_copt=external/picosat/picosat.c@-fno-sanitize=pointer-overflow
-
 ### Memcheck build. ###
 build:memcheck --build_tests_only
 build:memcheck --copt -g
diff --git a/tools/workspace/picosat/repository.bzl b/tools/workspace/picosat/repository.bzl
index 3ab4d1485d..0ee14d0a5c 100644
--- a/tools/workspace/picosat/repository.bzl
+++ b/tools/workspace/picosat/repository.bzl
@@ -11,8 +11,8 @@ def picosat_repository(
     github_archive(
         name = name,
         repository = "dreal-deps/picosat",  # License: MIT
-        commit = "4ee7aa1d1c645df8fa9daa07f2be17c6d03b35fc",  # v965
-        sha256 = "1be461d3659d4e3dc957a718ed295941c38dc822fd22a67f4cb5d180f0b6a7a3",  # noqa
+        commit = "a37ff03c5f78678f06801f501b54b1b04f662374",  # v965
+        sha256 = "e1d417c22c61b9d8344d4d5ad4132cf7bc25bb45e6434b97d6d134d1cb80fbb1",  # noqa
         build_file = "@dreal//tools:picosat.BUILD.bazel",
         mirrors = mirrors,
     )
```

I get the following output.

```
FAIL: //solvers:dreal_solver_test (see /media/sdb1/cache/bazel/_bazel_jwnimmer/10ee8aea36b37b216ef85df1451b64ad/execroot/drake/bazel-out/k8-dbg/testlogs/solvers/dreal_solver_test/test.log)
INFO: From Testing //solvers:dreal_solver_test:
==================== Test output for //solvers:dreal_solver_test:
Using drake_cc_googletest_main.cc

[==========] Running 50 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 50 tests from DrealSolverTest
[ RUN      ] DrealSolverTest.Interval
[       OK ] DrealSolverTest.Interval (0 ms)
[ RUN      ] DrealSolverTest.Available
[       OK ] DrealSolverTest.Available (0 ms)
[ RUN      ] DrealSolverTest.CheckSatisfiabilityTrivialUnsat
[       OK ] DrealSolverTest.CheckSatisfiabilityTrivialUnsat (0 ms)
[ RUN      ] DrealSolverTest.CheckSatisfiabilityTrivialSat
[       OK ] DrealSolverTest.CheckSatisfiabilityTrivialSat (0 ms)
[ RUN      ] DrealSolverTest.CheckSatisfiabilityConjunction
[       OK ] DrealSolverTest.CheckSatisfiabilityConjunction (0 ms)
[ RUN      ] DrealSolverTest.CheckSatisfiabilityDisjunction
dreal_solver_test: external/picosat/picosat.c:3138: void original_clauses_satisfied(PS *): Assertion `clause_satisfied (c)' failed.
```

@mwoehlke-kitware can you work on a proper patch, that passes Drake's unit tests?

rpoyner-tri, 2022-06-30:
Worked for me:
[0001-wip.patch.zip](https://github.com/RobotLocomotion/drake/files/9023489/0001-wip.patch.zip)

apply on top of PR#1.

jwnimmer-tri, 2022-06-30:
Open a Drake PR that adds your patch locally, like so:

```patch
--- a/tools/workspace/picosat/repository.bzl
+++ b/tools/workspace/picosat/repository.bzl
@@ -14,5 +14,8 @@ def picosat_repository(
         commit = "4ee7aa1d1c645df8fa9daa07f2be17c6d03b35fc",  # v965
         sha256 = "1be461d3659d4e3dc957a718ed295941c38dc822fd22a67f4cb5d180f0b6a7a3",  # noqa
         build_file = "@dreal//tools:picosat.BUILD.bazel",
+        patches = [
+            "@drake//tools/workspace/picosat:patches/ubsan.patch",
+        ],
         mirrors = mirrors,
     )
```

(Add your patch file in also, at that location.)

If the patch ever gets upstreamed, then we can drop it from Drake.

PR #17505 (2022-07-06): [build system] Update picosat to latest commit
Closes: #17139

Also drop disablement of UBSan checks specific to picosat; that is no
longer needed.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17505)
commit ed661c77 (2022-07-06): [workspace] Update picosat to latest commit (#17505)

* [workspace] Update picosat to latest commit

Closes: #17139

Also drop disablement of UBSan checks specific to picosat; that is no
longer needed, thanks to fixes upstream.
closed 2022-07-06 by rpoyner-tri