RobotLocomotion/drake #17149: Meshcat does not support capsule geometry.

JoseBarreiros-TRI, 2022-05-10:
Meshcat trows a warning and fails to render when visualizing a simple capsule defined in a sdf file. 
E.g. `bazel run //manipulation/util:show_model ~/anzu/models/ruxid/cylinders.sdf -- --meshcat default`
warning:
`home/josebarreiros/.cache/bazel/_bazel_josebarreiros/90f3e8e4e37bf5312f47e87299f7a9c9/execroot/drake/bazel-out/k8-opt/bin/manipulation/util/show_model.runfiles/drake/bindings/pydrake/systems/meshcat_visualizer.py:550: UserWarning: Unsupported shape <pydrake.geometry.Capsule object at 0x7fb3ba26d4f0> ignored
  warnings.warn(f"Unsupported shape {shape} ignored")`

jwnimmer-tri, 2022-05-10:
That's a message from the old, deprecated visualizer (see #16978).  Unless you have an example from the current code, we should close this as "won't fix".

jwnimmer-tri, 2022-05-10:
Okay, as best I can tell, this is a duplicate of #16978 -- we need to port show_model to use the new MeshcatVisualizer bindings.

Please comment or re-open in case there is a different problem I've overlooked.

JoseBarreiros-TRI, 2022-05-11:
@jwnimmer-tri It seems that the new meshcatvisualizer does not support capsule either.
when using:
`meshcat_vis = MeshcatVisualizerCpp.AddToBuilder(builder, scene_graph, meshcat)`

`WARNING:drake:Meshcat does not display Capsule geometry (yet).
WARNING:drake:Meshcat does not display Capsule geometry (yet).`

jwnimmer-tri, 2022-05-11:
That's true.  We can re-open this issue to track that request:

https://github.com/RobotLocomotion/drake/blob/bf5947f26b77b5c47d931814404de96d89ddca83/geometry/meshcat.cc#L305-L307

PR #17319 (2022-11-02): [meshcat] Add capsule support
Fixes #17149

Depends on rdeits/meshcat#116 before merging.

THREE.js added native capsule shape support in mrdoob/three.js#23586. This PR adds capsule support to the in process meshcat class and temporarily points to a fork of meshcat with an updated THREE.js version.

cc: @JoseBarreiros-TRI

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17319)
closed 2022-11-02 by xuchenhan-tri