RobotLocomotion/drake #17152: Feature request to add geometries with elliptical cross section

JoseBarreiros-TRI, 2022-05-10:
There is currently no way to create or compose non-symmetrical rounded geometries (e.g. with ellipsoidal cross section). It would be a nice feature to have. Some examples:

Ellipsoid
[image]

Ellipsoidal Cylinder 
[image]

Ellipsoidal Capsule

jwnimmer-tri, 2022-05-10:
Could you elaborate a bit more about what you're trying to do?

Drake supports arbitrary convex meshes (and non-convex meshes, under certain conditions), and it seems to me like convex meshes would be able to represent these shapes, as least insofar as I can understand them from the images above.

JoseBarreiros-TRI, 2022-05-10:
This request is the context of hydroelastics. My understanding is that if using a convex mesh, the resolution cannot be changed from drake. This results in a trial and error test of various resolution meshes created externally. On the other hand, if using primitive shapes (e.g. cylinder), we can change the resolution directly in the SDF. This advantage makes the primitive geometries preferred for contact modeling; however, the geometry palette is limited. Tagging @DamrongGuoy to confirm this.

jwnimmer-tri, 2022-05-10:
> ... when using a convex mesh, the resolution cannot be changed from drake ...

How about we fix _that_, instead?

Carefully hand-tuning primitive shapes to use as collision geometries is outdated; we shouldn't double-down on that broken workflow.  We should be able to simulate from a CAD-output mesh (or volume), with the ability to tesellate (or decimate) that volume, perhaps subject to the same kind of resolution hint that primitive shapes already offer.

sherm1, 2022-05-10:
I have a better idea -- use the Elllipsoid primitive!

closed 2022-05-10 by sherm1

jwnimmer-tri, 2022-05-10:
That covers Ellipsoid, but the OP also requested Elliptic Cylinder and Elliptic Capsule.

(I'm still happy with "wontfix" on that front, though. I don't think we should add so many more primitives.)

JoseBarreiros-TRI, 2022-05-10:
> Carefully hand-tuning primitive shapes to use as collision geometries is outdated; we shouldn't double-down on that broken workflow. We should be able to simulate from a CAD-output mesh (or volume), [...]

I agree with this, having the ability to remesh the volume would be really useful.