RobotLocomotion/drake #17160: Accelerate Polynomial Parse Time

lujieyang, 2022-05-11:
Currently, `Polynomial` takes a decent amount of time to parse `Expression ` as `Polynomial` when the degree gets high. Could we accelerate the process by incorporating parallelization? The following code snippet can reproduce the issue:

`from pydrake.all import (MathematicalProgram, Polynomial, Variables)
import numpy as np
import time

nz = 3
deg = 16

prog = MathematicalProgram()
z = prog.NewIndeterminates(nz, "z")
J = prog.NewFreePolynomial(Variables(z), deg)
J_expr = J.ToExpression()

dJdz = J_expr.Jacobian(z)
f2 = np.array([[0], [0], [1]])
u_opt = -0.5 * (f2.T).dot(dJdz.T)

z0 = np.array([0, 1, 0])
l = (z-z0).dot(z-z0) + u_opt.dot(u_opt)
f = np.array([
            z[1] * z[2],
            - z[0] * z[2],
            (z[0] + u_opt[0] -  z[2])])
RHS = l + dJdz.dot(f)
start_time = time.time()
Polynomial(RHS)
end_time = time.time()
print("Polynomial parse time: ", end_time-start_time)`

jwnimmer-tri, 2022-05-11:
For reference, on my Ubuntu 20 workstation when I run this against Drake v1.2.0, it reports a parse time of approximately 41 seconds.

hongkai-dai, 2022-05-12:
BTW. For that timed operation
```
Polynomial(RHS)
```
This line will treats both `z` and the coefficient of J as indeterminates.

I think you want
```
Polynomial(RHS, sym.Variables(z))
```
which treats only `z` as indeterminates.

hongkai-dai, 2022-05-12:
Another interesting fact: instead of constructing `RHS` as symbolic::Expression and then parse it as Polynomial, I tried the other approach to construct each expression as Polynomial and directly operate on these polynomials. This takes 280s on my machine. Here is the code snippet
```py
from pydrake.all import (MathematicalProgram, Polynomial, Variables)
import numpy as np
import time

nz = 3
deg = 16

start_time = time.time()
prog = MathematicalProgram()
z = prog.NewIndeterminates(nz, "z")
J = prog.NewFreePolynomial(Variables(z), deg)

dJdz = J.Jacobian(z)
f2 = np.array([[Polynomial()], [Polynomial()], [Polynomial(1)]])
u_opt = -0.5 * (f2.T).dot(dJdz.T)

z0 = np.array([0, 1, 0])
l = Polynomial((z-z0).dot(z-z0)) + u_opt.dot(u_opt)
f = np.array([
            Polynomial(z[1] * z[2]),
            Polynomial(- z[0] * z[2]),
            (Polynomial(z[0] - z[2]) + u_opt[0])])
RHS = l + dJdz.dot(f)
end_time = time.time()
print("Polynomial parse time: ", end_time-start_time)
```

jwnimmer-tri, 2022-05-13:
I think the next step here is to re-work the above code into a new test case in the C++ code for [benchmarking mathematical program](https://github.com/RobotLocomotion/drake/blob/master/solvers/benchmarking/benchmark_mathematical_program.cc).

I believe once the fixup in https://github.com/RobotLocomotion/drake/issues/17160#issuecomment-1124588816 is performed, the runtime goes beyond 60 seconds.  For the benchmark, we'll want to reduce `nz` or `deg` (or both) so that it runs in a reasonable amount of time for now (something around 30 seconds should be fine).

After that, we'll be able to more easily profile the program to see where the time is going.

hongkai-dai, 2022-05-13:
I will send a PR to add the benchmark, and we can work from there.

hongkai-dai, 2022-05-17:
kcachegrind shows that most of the time is spent on expanding the expression. Specifically `ExpressionAddFactory::AddTerm` takes a lot of time, especially on these two lines
https://github.com/RobotLocomotion/drake/blob/fdff101e6a8fdacbd2ea02b0ab257aa4b3be3551/common/symbolic_expression_cell.cc#L702
and
https://github.com/RobotLocomotion/drake/blob/fdff101e6a8fdacbd2ea02b0ab257aa4b3be3551/common/symbolic_expression_cell.cc#L718

each line takes more than 20% of runtime.

My first instinct is to change `expr_to_coeff_map_` from std::map to std::unordered_map to change the complexity from O(logN) to O(1). But I think that would be a bad idea as we loop over this `expr_to_coeff_map_` a lot. I will see if there are other fixes.

jwnimmer-tri, 2022-05-17:
I did look into the profiling a bit.  I think the reason those are expensive is not because the map is slow, but because we are calling them too many times because we're doing duplicate work while over-expanding.  The profile points us to `Expand()` (not `Polynomial()`), but after that I think we need more human ingenuity, not profiling.  If we work out the expression math by hand (what are the terms, where are products distributed and simplified, etc.), I think we will find a lot of places where we're doing unnecessary calculations.  The best way to speed up code is not to make it run faster, but rather to not run it at all.

jwnimmer-tri, 2022-05-17:
Basically, what I started doing (and you could do too) is to manually trace through this benchmark with a lower degree (e.g., 3 or something).  Write out the algebra by hand (dot products, etc.), keeping it in expression-tree form just like symbolic::Expression does.  Watch what the code does for that algebra.  Is it doing non-trivial work during `Expand()`, in cases where the expression is already expanded?  I definitely found a few, things like distributing `0.5*` across an Expression should not affect the expansion, but I think it was.

hongkai-dai, 2022-07-27:
With the recent improvement on symbolic Polynomial class and symbolic Expression, the polynomial parsing time has been sped up significantly. I will close this issue for now. We can open a new issue if the parsing time is slow again.

closed 2022-07-27 by hongkai-dai