RobotLocomotion/drake #17161: APT improvements: dependencies

svenevs, 2022-05-11:
Sub-component of #16448.  Relates: https://github.com/RobotLocomotion/drake/pull/17058#pullrequestreview-965130560 and https://docs.google.com/document/d/1ZFq8EWLAVKHoUpJaf-Qx1AWzXllmVYCB_W1JOi-TnCQ

The `.tar.gz` archives should have their setup installation scripts allow for:

- [ ] `shlibs:Depends` has `libblas3 | libblas.so.3`, we have `libblas3`.
    - The “|” form is allowing the user to choose a different implementation.
- [ ]  `shlibs:Depends` installs `libglu1-mesa | libglu1`, we install `libglu1-mesa`.
- [ ] `shlibs:Depends` installs `liblapack3 | liblapack.so.3`, we install `liblapack3`.
- [ ] We install `libldl2`, `shlibs:Depends` does not.
    - This may be a leftover (or statically linked now?), we should check.
- [ ] `shlibs:Depends` installs `libstdc++6 (>= 9)`, we do not.
    - It’s probable that we already get this implicitly, but we should double-check.
- [ ] `shlibs:Depends` installs `libtinyxml2-6a (>= 6.0.0)`, we install `libtinyxml2-dev` and `libtinyxml2.6.2v5`.
    - The `libtinyxml2.6.2v5` is probably wrong.
- [ ] We install `libyaml-cpp-dev`, `shlibs:Depends` does not.
    - This is a hold-over from some code that got removed from Drake.

svenevs, 2022-05-12:
Additional `apt` analysis for installed packages can be found here: https://gist.github.com/svenevs/6c9e247d7dc71f3f595182181cf099b7 (includes other information about files installed to `/opt/drake` that can be ignored for this as part of #17058).

jwnimmer-tri, 2023-01-20:
I skimmed through the list above; all of it seems on point.

> We install libldl2, shlibs:Depends does not.
> This may be a leftover (or statically linked now?), we should check.

I think it was probably an accident when we added #12072, and has never been used.  From suitesparse we only use `libamd2`, which doesn't need `libldl2` at all.

mwoehlke-kitware, 2023-02-02:
> we install ... `libtinyxml2.6.2v5`

We do? Where? (Did this get addressed but not checked off?)

jwnimmer-tri, 2023-02-02:
See #17830 re: tinyxml2.

mwoehlke-kitware, 2023-02-02:
"Drop tinyxml2 as of the deprecation 2023-01-01"

Okay, it's 2023-02-02 🙂 . So the action item for tinyxml is to remove `libtinyxml2-dev` also?

jwnimmer-tri, 2023-02-02:
Yes.  In general, any shared libraries that we don't dynamically load anymore in the current release shouldn't be part of `Depends:` in the control file.  There might be more newly-unused ones beyond tinyxml2.

mwoehlke-kitware, 2023-02-02:
Ugh. `libtinyxml2-dev` vs. `libtinyxml-dev`. I don't see the former. So, it looks like that item can be checked?

jwnimmer-tri, 2023-02-02:
The general idea is to match the Depends list (which is specified by `setup/ubuntu/binary_distribution/packages-{focal|jammy}.txt`) with the dynamic loader dependencies, possibly assisted by `dpkg_shlibsdeps` recommendations about things that are extra/missing.  You'll need to dig in, see what's what, and figure out any necessary changes or conclude that none are necessary.

mwoehlke-kitware, 2023-02-02:
Since I can't find any evidence that "we install `libtinyxml2-dev` and `libtinyxml2.6.2v5`", and all indications are that we used to do so and no longer do so, I'm marking that complete.

svenevs, 2023-02-07:
In order to evaluate the differences between what our `packages-*.txt` are
defining and what `shlibs:Depends` is doing requires modifying the repackaging
script and repackaging a `.tar.gz` archive.  This line is ultimate what defines the dependencies:

https://github.com/RobotLocomotion/drake/blob/777507402917549c299c7af17366fe7112e0404f/tools/release_engineering/repack_deb.py#L74

To compare, you have to overwrite it with

```diff
--- a/tools/release_engineering/repack_deb.py
+++ b/tools/release_engineering/repack_deb.py
@@ -72,6 +72,7 @@ def _run(args):
     # NOTE: rstrip is required here so that the last line does *NOT* have a
     # comma (otherwise debian/rules will crash).
     depends = packages_txt.rstrip().replace('\n', ',\n         ')
+    depends = "${shlibs:Depends}"
     with open(deb_control_in, encoding='utf-8') as f:
         deb_control_contents = f.read().format(depends=depends)
```

What this does is in the resultant `.deb` file, our "control" file gets updated
to use what `shlibs:Depends` thinks we should have instead of our current list.
Examining this is a little unpleasant (focal must be repackaged on focal,
jammy on jammy, so if your host is focal run a jammy docker and run drake's
setup scripts with `--with-maintainer-only`).  Once you can
`bazel run //tools/release_engineering:repack_deb -- --tgz /path/to/drake-thing.tar.gz --output-dir /abs/path/to/output` you will get a `.deb` file.  You then need
to extract the dependencies:

```console
# extract archives
$ ar x drake-dev_0.0.20230207185335-1_amd64.deb

# extract control file, focal: control.tar.xz, jammy: control.tar.zst
$ tar -xf control.tar.xz

# Dump the dependencies to a text file
$ cat control | /bin/grep Depends: | sed 's/Depends: //g' | tr ',' $'\n' | sed 's/^ //g' | sort > packages-focal.txt
```

When reviewing the contents of a PR, you can run a sample deb packaging job:

- https://drake-jenkins.csail.mit.edu/job/linux-focal-unprovisioned-gcc-bazel-experimental-snopt-mosek-packaging
- https://drake-jenkins.csail.mit.edu/job/linux-jammy-unprovisioned-gcc-bazel-experimental-snopt-mosek-packaging

When the job is complete search the logs (go to `/consoleFull`) for `Upload complete:`, where you can get
both the `.tar.gz` archive (that you can run through `repack_deb` after modifying `depends = '${shlibs:Depends}`)
as well as the `.deb` archive which will let you examine the existing control file.

----

I will now include the analysis after comparing the results using PR #18734 to
look at the remaining / new items that have come up.  The last time we went
through this there were some packages already cleared via
https://github.com/RobotLocomotion/drake/pull/17058#pullrequestreview-968050771
and the [related google doc](https://docs.google.com/document/d/1ZFq8EWLAVKHoUpJaf-Qx1AWzXllmVYCB_W1JOi-TnCQ/edit?usp=sharing).
Summary at the end of this comment with potential action items.

## focal

### Found in `${shlibs:Depends}` but not in `packages-*.txt`:

- [X] libblas3 | libblas.so.3 (expected, re: `apt-get satisfy`, tracked in #17161)
- [X] libc6 (>= 2.29) (OK, dismissed in #17058)
- [ ] libcrypt1 (>= 1:4.1.0) (Unclear, not addressed in #17161, we no longer have curl anyhere)
- [X] libgcc-s1 (>= 4.2) (OK, dismissed in #17058)
- [X] libgl1 (OK, dismissed in #17058)
- [X] libjpeg8 (OK, dismissed in #17058)
- [X] libqt5gui5 (>= 5.12.2) | libqt5gui5-gles (>= 5.12.2) (OK, ignore, it also lists a bare `libqt5gui5` which we have)
- [ ] libquadmath0 (unclear, usually comes from fortran right?)
- [X] libspdlog1 (OK, dismissed in #17058)
[... 90 lines cut ...]
- [X] libnetcdf19
- [X] libogg0
- [X] libxml2
- [X] libxt

-----------

Noting that this process is somewhat arduous, attached are the extracted `.txt`
documents that produced the above analysis for people to cross reference.
New packages to consider that `${shlibs:Depends}` wants:

- Potentially missing packages:
    - focal and jammy:
        - libcrypt1
        - libquadmath0
    - jammy only (?):
        - Seems likely this just comes from spdlog-dev / packaging differences
          OR from VTK tarball situation.  I'm 95% sure these can be ignored.
        - libfmt8
        - libspdlog1-fmt8
- Potentially extra packages:
    - default-jre
    - jupyter-notebook
    - libmsgpackc2
    - libnlopt-cxx0

jwnimmer-tri, 2023-05-18:
> Found in ${shlibs:Depends} but not in packages-*.txt:
> libquadmath0 (unclear, usually comes from fortran right?)

Our packages.text lists `libgfortran5` and `libgfortran5` depends on `libquadmath0` so this is no action required.  Shlibdeps is aggressive about transitive dependencies.  We do not need to be in packages.txt.

> libcrypt1 (>= 1:4.1.0)

This is `Priority: required` so there's no real need for packages.txt to care about it.  Presumably something we grab from Ubuntu us using it transitively.

jwnimmer-tri, 2023-05-18:
>jammy only (?):
>
>    Seems likely this just comes from spdlog-dev / packaging differences
>    OR from VTK tarball situation. I'm 95% sure these can be ignored.
>    libfmt8
>    libspdlog1-fmt8

For the same reason, these are fine already (no need to list them).  Our txt file says `libspdlog-dev` which pulls in those runtime libraries already.

jwnimmer-tri, 2023-05-18:
>Potentially extra packages:
> libmsgpackc2

Already fixed by #19080.

> default-jre

Not a shared library, so would not be a part of shlibdeps output.  We need this for `lcm-spy` and such.  No changes necessary.

> jupyter-notebook

Not a shared library, so would not be a part of shlibdeps output.  We need this for the tutorials.  No changes necessary.

> libnlopt-cxx0

Looks like #18532 forgot to clean this up.  I'll open a PR.

That's everything, so just that one change and I'll close this.

jwnimmer-tri, 2023-05-18:
Merged!

closed 2023-05-18 by jwnimmer-tri