RobotLocomotion/drake #17162: ipopt_solver_test: passes from source build, but not via pip build?

EricCousineau-TRI, 2022-05-11:
Relates https://stackoverflow.com/questions/72177582/drake-mathematical-program-tutorial
(but this is on Ubuntu 20.04, no WSL)

Take release `v1.2.0`, and the following test:
https://github.com/RobotLocomotion/drake/blob/v1.2.0/bindings/pydrake/solvers/test/ipopt_solver_test.py

In source, the test passes:
```
cd drake
git checkout v1.2.0  # no other changes
bazel run //bindings/pydrake/solvers:py/ipopt_solver_test
```

From pip:
```
cd $(mktemp -d)
python3 -m venv .
source ./bin/activate
pip install -U pip wheel
pip install drake==1.2.0
# add `unittest.main()` to the bottom of script
python .../ipopt_solver_test.py
```

I get following failure:
```
...
    self.assertTrue(result.is_success())
AssertionError: False is not true
```

Per StackOverflow post, I also try to print out `result.get_solver_details().status` before assert error, and I get the same error, `RuntimeError: The solver_details has not been set yet.`

I did not try anything w/ `tgz`, so not sure what source of issue is.

Assigning to Jeremy for disposition

jwnimmer-tri, 2022-05-11:
Our wheel build of Ipopt must be misconfigured or faulty in some way:

https://github.com/RobotLocomotion/drake/blob/bf5947f26b77b5c47d931814404de96d89ddca83/tools/wheel/image/dependencies/projects/ipopt.cmake#L1-L17

The version pin matches Ubuntu 20, so that's not it.  Maybe Ubuntu adds patches, or en/disables certain sub-solver options.  We can check the Ubuntu source package to find out.

jwnimmer-tri, 2022-05-12:
I think maybe we need to enable MUMPS when building Ipopt for the wheel.

jwnimmer-tri, 2022-05-25:
Running `ipopt_solver_test.py` linked in the OP (with a PYTHONPATH pointing at the wheel) will demonstrate the failure.

mwoehlke-kitware, 2022-05-25:
> Running `ipopt_solver_test.py` ...

That's a rather large chunk of code to add. Remember, that file doesn't exist in the testing environment. If we're going to start wholesale adding Python tests, we should probably go "all in" and add *all* of them, which is likely to be a non-trivial (but probably worthwhile) effort.

Otherwise, I'd prefer if there is something simpler that can be added to [`test-wheel.sh`](https://github.com/RobotLocomotion/drake/blob/master/tools/wheel/test/test-wheel.sh). Keep in mind that the current tests *are in a heredoc* in that script.

jwnimmer-tri, 2022-05-25:
> ... we should probably go "all in" and add all [tests] ...

That is probably worthwhile, but need not get in the way to doing something simple for this one, urgent ticket.

> That's a rather large chunk of code to add.

It's just one file, and is only 50 lines long?  It doesn't depend on anything except code already in the wheel.

> Remember, that file doesn't exist in the testing environment.

Sure, but that's up to us, it's not a rule of nature.  We could copy it into the test environment if we wanted to.

> ... something simpler that can be added  ...

As best I understand it, the check for `result.is_success()` would be a sufficient test condition to catch this bug.  If you'd like to scrape the relevant ~10 lines of the full test into a smaller wheel-specific test, that's fine by me too.  The goal would be a test that fails against the current wheel, but passes once the wheel is repaired.

Whatever we do for an automated test, we should still manually test that the reproduction recipe atop the ticket is fully resolved by the wheel fix.

PR #17346 (2022-06-07): Add MUMPS to IPOPT when building wheels
Point our wheel build of IPOPT at MUMPS. Because MUMPS is copyleft, we must consume it as a shared library. On Ubuntu, this means we can just use the distro-provided package. On macOS, we must build it from source, as it is not packaged separately in Homebrew. (Homebrew normally builds it as part of the IPOPT build.)

Fixes #17162.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17346)
closed 2022-06-07 by jwnimmer-tri