RobotLocomotion/drake #17167: Fix GetSolution for continuous variables x_v in GraphOfConvexSets

TobiaMarcucci, 2022-05-12:
Currently the retuned value for the variables `x_v` is `sum_{e in Eout_v} y_e` (see [here](https://github.com/RobotLocomotion/drake/blob/74422593c2741ed4bf8713ce5801aa7d95fdf622/geometry/optimization/graph_of_convex_sets.cc#L655)). While this is correct for the vertices along the optimal path in case of an integer solution, it can be misleading when solving the convex relaxation.

The following two options recover the assignment above in case of an integer solution, and return a point `x_v` that is guaranteed to be in the set `X_v` in case of fractional flows:
- `x_v := sum_{e in Eout_v} y_e / sum_{e in Eout_v} phi_e` if `phi_e > 0` for at least one edge outgoing `v`,  and `x_v := center_of_X_v` otherwise.
- `x_v := sum_{e in Eout_v} y_e + (1 - sum_{e in Eout_v} phi_e) * center_of_X_v`.

I slightly favor the first, although the second does not require to check if the flows are "close enough" to zero. In case of the target vertex `t` the definitions above need to sum over the incoming edges and use `z_e` in place of `y_e`.

RussTedrake, 2022-05-14:
We do not have center() implemented for all convex sets, and for HPolyhedron, it requires solving and optimization.  I would think that's too expensive for filling in the values of the vertices that nobody should be using?

I would propose `x_v := sum_{e in Eout_v} y_e / sum_{e in Eout_v} phi_e` if `phi_e > tol` for at least one edge outgoing `v`, and `x_v := 0` otherwise; and documenting it as such in the method documentation.  Unfortunately, setting `tol` correctly probably depends on the solver tolerances -- I don't have a great proposal for that.

RussTedrake, 2022-05-14:
cc @mpetersen94

TobiaMarcucci, 2022-05-14:
I agree on avoiding the computation of the center. For the case where the flow is zero, would something like a suitably-sized vector of `nan`s or the equivalent of python's `None` be more interpretable from a user perspective (instead of `x_v := 0`)?

Random idea regarding the tolerance. One could try to always do the division (except, of course, when `phi_e` is exactly zero). Then check if the rescaled point is in the convex set `X_v`. If yes, return it. If not, return zero (or `nan`s or `None`). Not sure how elegant this is, but it'd prevent us from picking a tolerance, and would always return something sensible.

RussTedrake, 2022-05-14:
I like that.  It will have to be `nan` since the solutions is a vector of doubles.  I will PR it now.

RussTedrake, 2022-05-15:
I think our plan does not work.  The projected point may not be in the set with tolerance zero due to solver precision.  Take the extreme case of the sets being a Point (which happens to be covered by a number of our unit tests).  The projected point is almost certainly not exactly equal to the desired point.  I actually think we cannot check PointInSet for any of the return values without some understanding of the solver tolerances.

I've implemented instead
```
    if (convex_relaxation) {
      if (sum_relaxed_phi < 100.0 * std::numeric_limits<double>::epsilon()) {
        x_v = VectorXd::Constant(v->ambient_dimension(),
                                 std::numeric_limits<double>::quiet_NaN());
      } else {
        x_v /= sum_relaxed_phi;
      }
    }
```
which I'm happy with.  I will open the PR now and we can discuss it there.

commit 8422edba (2022-05-17): [geometry/optimization] Fix GetSolution for x_v in GraphOfConvexSets (#17199)

* [geometry/optimization] Fix GetSolution for x_v in GraphOfConvexSets

Resolves #17167
closed 2022-05-16 by RussTedrake