RobotLocomotion/drake #17170: geometry: Should expose public API for `add_triad` for frame visualization

EricCousineau-TRI, 2022-05-12:
Per https://github.com/robotlocomotion/drake/pull/17141#issuecomment-1121847796

@JoseBarreiros-TRI is this something you'd be up for? (at low priority)

adamconkey, 2023-04-06:
Has the status of this issue changed any? I realized I've been using the `AddTriad` function from the [manipulation repo](https://github.com/RussTedrake/manipulation/blob/master/manipulation/scenarios.py#L506-L563), but it'd be great if this was in the Drake API instead.

jwnimmer-tri, 2023-04-06:
What's changed is that the code is part of `pydrake` now (not buried inside Bazel):

https://github.com/RobotLocomotion/drake/blob/98aa589903abd1585b8978a24abc70b26cb73493/bindings/pydrake/visualization/_model_visualizer.py#L488

On the other hand, it's still private and not terribly easy to reuse.

Somehow this issue never hit my radar.  I'll see if I can take a quick spin to improve it.

adamconkey, 2023-04-06:
I see, thanks! So maybe I can hack it by creating a new `ModelVisualizer` instance and manually setting `self._builder` with the builder I've already created, and then call `_add_triad`.

jwnimmer-tri, 2023-04-06:
Yeah, there's surely some way to hack it.  It's probably pretty gross though.  I think I should be able to make a public function before the next pydrake release.

adamconkey, 2023-04-06:
What I described did work, but you're right, it's ugly. A public would be much appreciated.

As an aside, this was the first I'm learning of `RobotDiagramBuilder`, I was separately creating the diagram builder, parser, plant, and scene graph. This is a nice convenience!

jwnimmer-tri, 2023-04-06:
WIP at https://github.com/RobotLocomotion/drake/pull/19152

PR #19152 (2023-04-10): [visualization] Add frame (aka triad) visualization function
Closes #17170.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19152)
closed 2023-04-10 by jwnimmer-tri