RobotLocomotion/drake #17177: Run deb repacker during Nightly Packaging, push to S3

svenevs, 2022-05-12:
Run [`repack_deb.py`](https://github.com/RobotLocomotion/drake/blob/aaf4caa3227607a85295ecb6d3297e053df7ba1b/tools/release_engineering/dev/repack_deb.py) as part of [linux-focal-unprovisioned-gcc-bazel-nightly-snopt-mosek-packaging](https://drake-jenkins.csail.mit.edu/view/Nightly/job/linux-focal-unprovisioned-gcc-bazel-nightly-snopt-mosek-packaging/) and upload the `.deb` artifacts to S3.

svenevs, 2022-05-12:
> We will probably want to defer that work until after the tool is moved out of dev folder?
>
> https://github.com/RobotLocomotion/drake/issues/16448#issuecomment-1125289072

@jwnimmer-tri I misread that / got the wires crossed a little bit from the meeting.  I was under the impression we wanted to have nightlies running to help stress test it before moving out of dev.  I started the prototype but can pick it up at a later date if preferred.

jwnimmer-tri, 2022-05-12:
For stress testing, I think using it manually for v1.3.0 will be sufficient.  That's the use case which we care most about.

I don't necessarily mind running it nightly now, but it would mean hard-coding a "dev" path into the CI scripts, which we'll need to flag-day change later.  I think we can use the tool next Monday or Tuesday as a sanity check, then platform review it middle next week, then turn it on in CI end of next week or early the following week?

svenevs, 2022-05-12:
Sounds good, platform review it post-release makes sense to me.  I'll continue preparations, I anticipate a bit of trial and error to get the S3 uploads working but we'll merge it in `drake-ci` after moving `repack_deb.py` out of dev.

svenevs, 2022-05-13:
Thoughts on where to upload things?  The official releases go to `drake-apt/{codename}/pool/main/d/drake/drake-dev_{version}.deb`.  Some options:

1. In `drake-apt/{codename}/pool`, create `experimental` and `nightly` folders.  Then upload to `d/drake/*.deb`.
2. Just create `drake-apt/{codename}/experimental/*.deb` and `drake-apt/{codename}/nightly/*.deb`.  Don't mess around with the `pool`s, since I don't think we're going to be updating all other relevant metadata in all the other folders.
3. Rather than `drake-apt`, upload them to `drake-packages` instead?  Meaning only the official releases go in `drake-apt`?
4. ???

Related, currently with `.tar.gz` the experimental jobs just upload `.tar.gz`, whereas nightly uploads `.tar.gz` and a `.sha512`.  Do we want to do the same thing here?  To my recollection we aren't going to sign experimental / nightly, so the `.sha512` for nightly would be consistent with the other nightly releases.

svenevs, 2022-07-12:
As of https://github.com/RobotLocomotion/drake-ci/pull/157 deb packaging (focal) is underway:

- https://drake-packages.s3.amazonaws.com/drake/nightly/drake-dev_0.0.20220712-1_amd64-focal.deb
- https://drake-packages.s3.amazonaws.com/drake/nightly/drake-dev_latest-1_amd64-focal.deb

Jammy builds will commence shortly (relates: https://github.com/RobotLocomotion/drake/issues/16217#issuecomment-1178416097 and #17546), nothing needs to change for CI utilities as far as I know (just promote jammy packaging to nightly once it is fixed).

closed 2022-07-12 by svenevs