RobotLocomotion/drake #17179: Universall and Ball joint parsed from a SDF do not create actuators when effort>0 and ignore the limits

JoseBarreiros-TRI, 2022-05-12:
Universall and Ball joint parsed from a SDF do not create actuators when effort>0. 
In addition, the limits are ignored. There is an SDF for testing:
https://github.com/JoseBarreiros-TRI/drake/blob/e19141ff36a102d52cef29f10dc3424a39cbb853/examples/ruxid/python_humanoid_sim/models/humanoid_v1_copy.sdf

jwnimmer-tri, 2022-05-12:
I started #17178 for giving out warnings, but I think it probably doesn't give enough warnings yet.  I'll investigate.

jwnimmer-tri, 2022-05-14:
As of the latest draft of #17178, we have these warnings now:

```
WARNING:drake:warning: Actuation (via non-zero effort limits) for ball joint 'neck_joint' is not implemented yet and will be ignored.
WARNING:drake:warning: Actuation (via non-zero effort limits) for universal joint 'shoulderL_joint' is not implemented yet and will be ignored.
WARNING:drake:warning: Actuation (via non-zero effort limits) for ball joint 'torso_joint' is not implemented yet and will be ignored.
WARNING:drake:warning: Actuation (via non-zero effort limits) for universal joint 'hipL_joint' is not implemented yet and will be ignored.
WARNING:drake:warning: Actuation (via non-zero effort limits) for ball joint 'ankleL_joint' is not implemented yet and will be ignored.
```

PR #17178 (2022-05-16): [parsing] Add warning for unsupported joint actuators
Closes #17179.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17178)
closed 2022-05-16 by jwnimmer-tri