RobotLocomotion/drake #17185: Ubuntu 22.04 ("Jammy") CI configuration

jwnimmer-tri, 2022-05-13:
Towards #16217, we'd like to define some Experimental builds to help prepare Ubuntu 22 support.

Let's add add definitions of (Provisioned & Unprovisioned) Experimental CI builds for Jammy (Release, Debug, Everything).  For now, no sanitizers, no wheels, no documentation, no packaging, etc.

BetsyMcPhail, 2022-05-13:
@jwnimmer-tri do we want to start with just unprovisioned jobs or add some provisioned ones as well?

jwnimmer-tri, 2022-05-13:
My thought was that it will take us a few cracks at the jammy `install_prereqs.sh` before it's in good shape, and so baking it into a provisioned AMI prior to that point would be a lot of churn with little gain.

BetsyMcPhail, 2022-05-13:
That makes sense to me, we'll start the process with the unprovisioned jobs.

svenevs, 2022-05-27:
drake-ci related focal specific components as of https://github.com/RobotLocomotion/drake-ci/commit/1b7dbf782367fa6f24994e2823927df034877d44 (note: drake itself has many more locations that may need to get fixed before CI works [#16931 and https://github.com/RobotLocomotion/drake/search?q=remove+bionic&type=commits], just whittling drake-ci here)

- [X] `./driver/configurations/bazel/step-build-docker-image.cmake:97:  if(DASHBOARD_UNIX_DISTRIBUTION_CODE_NAME STREQUAL "focal")`
    - Special builds are focal: https://github.com/RobotLocomotion/drake-ci/pull/145/files#diff-2d973e8044b1284915d415c42530a1ae6efd8c80381f2cba4db5999b4101dd37
- [X] `./driver/configurations/bazel/step-push-docker-image.cmake:86:  if(DASHBOARD_UNIX_DISTRIBUTION_CODE_NAME STREQUAL "focal")`
    - Special builds are focal: https://github.com/RobotLocomotion/drake-ci/pull/145/files#diff-1ab1094b049dd9cd165fa0302ba8ba58af27f1be8147adacad414cc2cc05a4e4
- [X] `./driver/configurations/bazel.cmake:324:  if(DISTRIBUTION STREQUAL "focal" AND DASHBOARD_TRACK STREQUAL "Nightly")`
    - Special builds are focal: https://github.com/RobotLocomotion/drake-ci/pull/145/files#diff-0f3946589d1f0c5e878795a90630276440176313b66a3c57a8011838bd05946c
- [ ] `./driver/environment.cmake:77:  set(DISTRIBUTION_REGEX "(focal)")`
    - Update to `(focal|jammy)`: 
- [X] `./driver/environment.cmake:197:  if(DISTRIBUTION STREQUAL "focal" AND TRACK STREQUAL "nightly")`
    - Docs builds remain on focal: https://github.com/RobotLocomotion/drake-ci/pull/134/files#diff-14f810998493045bbaa7aff9ea6cdd5bbf593768b17476aa4f719ec2aed3cdd4
- [X] `./driver/environment.cmake:208:  if(DISTRIBUTION STREQUAL "focal" AND TRACK STREQUAL "continuous")`
    - Mirror to S3 remains focal.  This one might change (?): https://github.com/RobotLocomotion/drake-ci/pull/134/files#diff-14f810998493045bbaa7aff9ea6cdd5bbf593768b17476aa4f719ec2aed3cdd4
- [ ] `./driver/environment.cmake:247:string(REGEX MATCH "^linux-focal-clang-bazel-(continuous|experimental|nightly)-leak-sanitizer$" REGEX_MATCH_RESULT "${DASHBOARD_JOB_NAME}")`
    - Caching right now is only for focal: https://github.com/RobotLocomotion/drake-ci/pull/136/files#diff-14f810998493045bbaa7aff9ea6cdd5bbf593768b17476aa4f719ec2aed3cdd4
- [ ] `./driver/environment.cmake:252:string(REGEX MATCH "^linux-focal-gcc-bazel-(continuous|experimental|nightly)-debug$" REGEX_MATCH_RESULT "${DASHBOARD_JOB_NAME}")`
    - Caching right now is only for focal: https://github.com/RobotLocomotion/drake-ci/pull/136/files#diff-14f810998493045bbaa7aff9ea6cdd5bbf593768b17476aa4f719ec2aed3cdd4
- [X] `./deb_testing/drake-latest-focal.deb.sha512:1:d784615fa27f8090e86720ab28e150aae02da5a25934bba350ba7d1ef5a3d353ddd568d7b2ac7628fd34c9c157009250d98a1685b939951709f0171309fc978d  drake-latest-focal.deb`
    - Does not seem relevant for this work.
- [X] `./run_local.bash:46:export JOB_NAME="linux-focal-gcc-bazel-experimental-release"`
    - Does not seem relevant for this work.

Job name detection regex seems to be the main blocker, related points of discussion being:

1. Eventually we may desire to do special builds for jammy, in which case some of the above may need modifications.
2. Eventually we may desire to do build caching on jammy.  That might be a short term desire, so we'd need to set that up and update the regex to `^linux-(focal|jammy)-...`?

jwnimmer-tri, 2022-05-27:
Jammy will have fewer builds defined in drake-jenkins-jobs than Focal for a quite while (no sanitizers etc, per above).

It's fine if we want to prepare the drake-ci "dashboard" jammy regexes extra jobs now, even if drake-jenkins-jobs won't exercise all of them yet.  There's no general reason we need to have anywhere that's just hard-coded "focal" instead of allowing for both.

Once we have Provisioned Jammy builds, we will want to turn on caching for those promptly.

jwnimmer-tri, 2022-06-07:
I recall a discussion last week about what work remains to do here, but I don't recall for sure what we want.  Could you transcribe into the issue here what remains to do?

svenevs, 2022-06-07:
Sure, sorry.  That's on my list for first thing tomorrow, I need t

- [ ] Confirm the legitimacy of the Drake CI google doc instructions on what _exactly_ is required for creating a new unprovisioned _linux_ image (or update them)
- [ ] Rebuild the jammy unprovisioned image, AMIs cannot be renamed and right now we're limping along on a test-tagged name that does not match the normal drake AMI naming scheme.

svenevs, 2022-06-08:
AMI rebuilt / S3 cleaned up, and docs updated.  To the best of our knowledge it is working as expected, but the jammy builds are littered with other issues that may be hiding potential image issues.  [gcc-release-everything](https://drake-jenkins.csail.mit.edu/view/Linux%20Jammy%20Unprovisioned/job/linux-jammy-unprovisioned-gcc-bazel-experimental-everything-release/5/console) and [clang-release](https://drake-jenkins.csail.mit.edu/view/Linux%20Jammy%20Unprovisioned/job/linux-jammy-unprovisioned-clang-bazel-experimental-release/6/console) logs seem to indicate the unprovisioned image is configured correctly.

I am going to consider this done, but there is one point of consideration -- a "problem" that's been around for a while unrelated to this ticket: over time, the unprovisioned and provisioned linux images end up running different kernel versions since unprovisioned does not `apt-get upgrade`.  So although an unprovisioned build would end up downloading and installing the newer kernel, it won't be running for that build.  Provisioned, on the other hand, would be running the newer kernels as we continue updating/upgrading those over time.  The only thing I see that realistically affecting would be any packages that do any udev rules or what have you like librealsense, but it's not clear that this is even relevant or worrisome for CI (no cameras are connected and running).

closed 2022-06-08 by svenevs