RobotLocomotion/drake #17189: Parametrizing `ConstraintRelaxingIk` to specify max num

5yler, 2022-05-13:
Context: 
- I'm using `ConstraintRelaxingIk` when planning online during robot operation in a context where speed is important. 
- The current implementation has [a few hardcoded parameters](https://github.com/RobotLocomotion/drake/blob/88fea982f7e7c10d1a809cf73c1c90a14719b8ba/manipulation/planner/constraint_relaxing_ik.cc#L58-L64) including `kMaxNumConstraintRelax`.
- IK calls which fail to find a solution take a much longer time to return vs. a successful IK call.

I'm looking into rewriting the class in a way that would:
- load the linked parameters above instead of hardcoding, and/or 
- allow a timeout value to be specified for the computation (regardless of whether or not the IK call succeeds)

Wondering if anyone has looked into this problem already, and/or has thoughts on the best way to structure the above.

hongkai-dai, 2022-05-15:
@siyuanfeng-tri as the original author of ConstraintRelaxingIk, could you take a look at this issue? Thanks!

5yler, 2022-05-15:
As a simple workaround I just added a `const int64_t timeout_ms` argument with a default value to the `PlanSequentialTrajectory` function, and break out of the while loop if the timeout is exceeded. I think making it optional in that way takes care of any reverse compatibility issues but also gives greater control to the caller on when to fail out of trying to solve.

5yler, 2022-05-15:
Alternatively: I tried putting together a derived class of `ConstraintRelaxingIk` with an overridden `PlanSequentialTrajectory` function but ran into issues with things being declared `private` - could these be changed to `protected`? I'm happy to PR either change but adding a timeout argument would be an actual API change to `PlanSequentialTrajectory`

jwnimmer-tri, 2022-05-16:
It would be very easy change those constants into parameters.  Something like this:
 
```c++
struct ConstraintRelaxingIkParams {
  int max_num_initial_guess = 50;
  int max_num_constraint_relax = 10;
  Vector3d initial_pos_tolerance(0.01, 0.01, 0.01);
  double initial_rot_tolerance = 0.01;
  double constraint_shrink_factor = 0.5;
  double constraint_grow_factor = 1.5;
};
```

Then for the user to specify them, we could have some combination of:

(1) Optionally pass `ConstraintRelaxingIkParams` to the constructor.

(2) Call `ConstraintRelaxingIkParams::SetParams(ConstraintRelaxingIkParams)`.

(3) Optionally pass `ConstraintRelaxingIkParams` to `PlanSequentialTrajectory`.

Would that be enough to resolve the request?

If you actually need a user-defined customization (e.g., timeout), then we would probably do that by providing a callback argument on `PlanSequentialTrajectory`, so the user can do as they like.  In your case, you could have your callback check the wall clock and tell the planner to cancel in case it's been too long.

jwnimmer-tri, 2022-05-16:
Update from offline chat -- simply offering `ConstraintRelaxingIkParams` will not be good enough; we basically want to offer cancellation in this API, via some kind of callback (or future/signal).

PR #17224 (2022-05-19): [planner] Add a keep_going option to ConstraintRelaxingIk
Closes #17189.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17224)
closed 2022-05-19 by EricCousineau-TRI