RobotLocomotion/drake #17203: Need API to check if solvers are available in pydrake

RussTedrake, 2022-05-15:
I just found myself (in pydrake) hoping to write something like:
```
if SolverAvailableFor(ProgramType.kMiSOCP):
```
but right now, it seems the best I can do is something more like:
```
mip_solver_available = True
try:
    MakeFirstAvailableSolver([MosekSolver.id(), GurobiSolver.id()]) is not None
except:
    mip_solver_available = False
```
Yuck!

This leads me to a few questions/requests:
- [ ] The `is_available` and `is_enabled` methods are not bound yet.  Can we bind them?
- [ ] Could/should we have `ChooseBestSolver` that works on `ProgramType` instead of `MathematicalProgram`?  It seems close.
- [ ] Could we make `ChooseBestSolver` or some variant return `std::optional<SolverId>` instead of throw?

I can break these requests into separate issues if that's more useful, but I would like to garner support for the right solution(s) first.

jwnimmer-tri, 2022-05-15:
> The `is_available` and `is_enabled` methods are not bound yet. Can we bind them?

A version of those are bound in the [base class](https://drake.mit.edu/pydrake/pydrake.solvers.mathematicalprogram.html#pydrake.solvers.mathematicalprogram.SolverInterface.available), as in `MosekSolver().available()`:

https://github.com/RobotLocomotion/drake/blob/2f6d0ddaf7c929a73fdc0a888a2e9286e002aa00/bindings/pydrake/solvers/test/mosek_solver_test.py#L19

We could bind the static functions as well, but it was less boilerplate to just skip them.  If you can demonstrate a use for the static functions as sugar, we can eat the boilerplate.  So far in my experience, iterating over a list of `Solver` object has been easy enough -- it's just an extra pair of `()`.

Possibly another thing to fix here would be to have the functions appear in the https://drake.mit.edu/pydrake/pydrake.solvers.mosek.html#pydrake.solvers.mosek.MosekSolver list of functions?  By default, it doesn't list inherited methods, but here maybe it should.

> Could/should we have ChooseBestSolver that works on ProgramType instead of MathematicalProgram? It seems close.

For a given `ProgramType`, some solvers can handle _some_ programs of that type, but not all of them (depending on the details of the program).  That's why solver selection uses the whole `MathematicalProgram` object.

I suppose we could make a `ChooseBestSolver(ProgramType)` that only looked at the type, but maybe better would be to do `vector<SolverId> GetAvailableSolvers(ProgramType)` that returned the whole list, so the user can still proceed in case the first (best) one cannot handle a particular program instance at runtime, and so that it's easy to return "None" (via empty), instead of throwing.

> Could we make `ChooseBestSolver` or some variant return `std::optional<SolverId>` instead of throw?

This seems fair enough, though it doesn't seem to address the `mip_solver_available = ...` use case?

RussTedrake, 2022-05-15:
Thanks.  I did look for those methods but somehow missed them.  The is_available / is_enabled and available / enabled dance is not my favorite.

I would think that ProgramType => ProgramAttributes => Solver support.  I agree that users can assume they will have a certain program type and get it wrong, but could you give an example of the gap you mentioned here?

jwnimmer-tri, 2022-05-15:
> Could you give an example of the gap you mentioned here?

The `EqualityConstrainedQPSolver` cannot handle `ProgramType == kQP` when the program contains a `ProgramAttribute::kLinearConstraint`.  There is no `ProgramType` which matches what `EqualityConstrainedQPSolver` can do.

The `UnrevisedLemkeSolver` cannot handle `ProgramType == kLCP` unless the constraints are an exact covering (at most once) of every decision variable.

jwnimmer-tri, 2022-05-15:
> Thanks. I did look for those methods but somehow missed them.

I think we agree now that this is not pydrake-specific?  We could drop "pydrake" from the title and `component: pydrake`?

hongkai-dai, 2022-05-16:
Thanks Jeremy.

It seems EqualityConstrainedQPSolver is an exception, that for all other solvers, we can determine if the solver can or cannot solver a certain program based on ProgramType. If that is true, I would vote that for a QP, we just don't return EqualityConstrainedQPSolver, namely
```
std::vector<SolverId> GetAvailableSolvers(ProgramType)
```
to return Gurobi, Mosek, etc but not EqualityConstrainedQPSolver for kQP. To be honest I almost never use EqualityConstrainedQPSolver in practice as it requires very specific type of constraints.

jwnimmer-tri, 2022-05-16:
It's okay by me if `vector<> GetAvailableSolvers(ProgramType)` is documented to be conservative, where it will only return solvers that can _definitely_ accept all programs of that type, even if there's a solver that might work for some subset of those types that will not be returned.  That's a benefit of that signature, versus `ChooseBestSolver(ProgramType)` where it's difficult to fulfill the contract given only the program type.

jwnimmer-tri, 2023-09-23:
Fixed in #17225.

closed 2023-09-23 by jwnimmer-tri