RobotLocomotion/drake #17237: Currently packaged mosek core dumps (sometimes) on Deepnote

RussTedrake, 2022-05-20:
The problem.  After [uploading and associating a license file](https://deepnote.com/workspace/russtedrake-b58d8141-6691-43e8-ad76-0575b319c8b8/project/Enabling-licensed-solvers-on-Deepnote-7581ca0c-04be-4632-9108-f0426b6c3fbb/%2Flicensed_solvers_deepnote.ipynb) on Deepnote, the following code is sufficient to crash the notebook kernel and generates a core file:
```
from pydrake.solvers.mathematicalprogram import MathematicalProgram
from pydrake.solvers.mosek import MosekSolver

prog = MathematicalProgram()
mosek = MosekSolver()
result = mosek.Solve(prog)
```
The error generated is 
```
Core was generated by `/usr/local/bin/python -m ipykernel_launcher -f /root/.local/share/jupyter/runti'.
Program terminated with signal SIGFPE, Arithmetic exception.
#0  0x00007f18a1210b59 in ?? () from /opt/drake/lib/libmosek64.so.9.2
```
Running this using a Deepnote terminal under `gdb python3`, I can see the the instruction that causes the fault is indeed a div
```
│  >0x7ffff2054b59      div    %ecx                                                                               
│   0x7ffff2054b5b      mov    %eax,0xda8(%rsp)                                                                   
│   0x7ffff2054b62      test   %r13d,%r13d                                                                        
│   0x7ffff2054b65      je     0x7ffff2054d57                                                                     
│   0x7ffff2054b6b      lea    0xd7302e(%rip),%rsi        # 0x7ffff2dc7ba0                                        
│   0x7ffff2054b72      xor    %eax,%eax                                                                          
│   0x7ffff2054b74      mov    0xda0(%rsp),%rdi                                                                   
│   0x7ffff2054b7c      mov    0xda8(%rsp),%edx  
```
with the following backtrace:
```
#0  0x00007ffff2054b59 in ?? () from /opt/drake/lib/libmosek64.so.9.2
#1  0x00007ffff192af25 in ?? () from /opt/drake/lib/libmosek64.so.9.2
#2  0x00007ffff1776f14 in ?? () from /opt/drake/lib/libmosek64.so.9.2
#3  0x00007ffff1711e91 in ?? () from /opt/drake/lib/libmosek64.so.9.2
#4  0x00007ffff17cbe30 in ?? () from /opt/drake/lib/libmosek64.so.9.2
#5  0x00007ffff17cbb93 in ?? () from /opt/drake/lib/libmosek64.so.9.2
#6  0x00007ffff17cd14f in MSK_optimize () from /opt/drake/lib/libmosek64.so.9.2
#7  0x00007ffff17ccdfd in MSK_optimizetrm () from /opt/drake/lib/libmosek64.so.9.2
```

This failure seemed to happen consistently in some notebooks on some days, and not on others.  I have now managed to have two Deepnote projects open at the same time, one failing and one passing.  

This one works:
```
root@deepnote:~/work # lscpu
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   46 bits physical, 48 bits virtual
CPU(s):                          2
On-line CPU(s) list:             0,1
Thread(s) per core:              2
Core(s) per socket:              1
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           85
Model name:                      Intel(R) Xeon(R) Platinum 8259CL CPU @ 2.50GHz
Stepping:                        7
CPU MHz:                         2970.788
BogoMIPS:                        4999.99
Hypervisor vendor:               KVM
Virtualization type:             full
L1d cache:                       32 KiB
L1i cache:                       32 KiB
L2 cache:                        1 MiB
L3 cache:                        35.8 MiB
NUMA node0 CPU(s):               0,1
Vulnerability Itlb multihit:     KVM: Vulnerable
Vulnerability L1tf:              Mitigation; PTE Inversion
Vulnerability Mds:               Vulnerable: Clear CPU buffers attempted, no microcode; SMT Host state unknown
Vulnerability Meltdown:          Mitigation; PTI
Vulnerability Spec store bypass: Vulnerable
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Full generic retpoline, STIBP disabled, RSB filling
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mm
                                 x fxsr sse sse2 ss ht syscall nx pdpe1gb rdtscp lm constant_tsc rep_good nopl xtop
                                 ology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq ssse3 fma cx16 pci
                                 d sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand h
                                 ypervisor lahf_lm abm 3dnowprefetch invpcid_single pti fsgsbase tsc_adjust bmi1 av
                                 x2 smep bmi2 erms invpcid mpx avx512f avx512dq rdseed adx smap clflushopt clwb avx
                                 512cd avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves ida arat pku ospke
```

and this one fails
```
lscpu
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   48 bits physical, 48 bits virtual
CPU(s):                          2
On-line CPU(s) list:             0,1
Thread(s) per core:              2
Core(s) per socket:              1
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       AuthenticAMD
CPU family:                      23
Model:                           1
Model name:                      AMD EPYC 7571
Stepping:                        2
CPU MHz:                         2545.941
BogoMIPS:                        4399.98
Hypervisor vendor:               KVM
Virtualization type:             full
L1d cache:                       32 KiB
L1i cache:                       64 KiB
L2 cache:                        512 KiB
L3 cache:                        8 MiB
NUMA node0 CPU(s):               0,1
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Vulnerable
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Full AMD retpoline, STIBP disabled, RSB filling
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mm
                                 x fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_g
                                 ood nopl nonstop_tsc cpuid extd_apicid aperfmperf tsc_known_freq pni pclmulqdq sss
                                 e3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand hypervisor lahf_l
                                 m cmp_legacy cr8_legacy abm sse4a misalignsse 3dnowprefetch topoext vmmcall fsgsba
                                 se bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv1 c
                                 lzero xsaveerptr arat npt nrip_save
```

So my current best hypothesis is that Mosek 9.2.33 which we current "ship" has a bug on AMD.  I will see if I can try some other mosek versions.

RussTedrake, 2022-05-20:
Good news(?).  Running under `!pip install Mosek==9.2.33`, the following code fails as described above:
```
import mosek

with mosek.Env() as env:
    with env.Task(0, 0) as task:
        task.optimize()
```

But running under `!pip install Mosek==9.3.20`, it succeeds.  So, although I would really like to use Mosek 10 (#17026), that is still in beta, so I think the solution is to upgrade Mosek to 9.3.20.

jwnimmer-tri, 2022-05-21:
Should this issue be closed now as of #17238?

RussTedrake, 2022-05-21:
Oops.  Yes.

closed 2022-05-21 by RussTedrake