RobotLocomotion/drake #17239: [geometry/hydroelastic] very small resolution hint causes overflow and incorrect mesh

joemasterjohn, 2022-05-20:
Per this [stack overflow post](https://stackoverflow.com/questions/72299376/the-hydroelastic-contact-does-not-give-expected-contact-surface/72307178#72307178) with the corresponding [reproduction branch](https://github.com/CWEzio/hydroelastic_issue).

For the cylinder mesh, a resolution hint that is too small will cause an overflow bug on [this line](https://github.com/RobotLocomotion/drake/blob/b4070eacafdc3fbeab2cb83e080f2e2917bcb888/geometry/proximity/make_cylinder_mesh.cc#L366). This likely occurs with our other shape types as well. We should just clamp to a minimum resolution hint (given the cylinder radius, etc.) and warn the user when they've given a hint that is too small.

SeanCurtis-TRI, 2022-06-03:
Looking at the sphere mesh generator, I'd point out two things:

1. The sphere mesh has the [same problem shown here](https://github.com/RobotLocomotion/drake/blob/b4070eacafdc3fbeab2cb83e080f2e2917bcb888/geometry/proximity/make_sphere_mesh.h#L581-L584), as you suggested.
2. A [few lines lower](https://github.com/RobotLocomotion/drake/blob/b4070eacafdc3fbeab2cb83e080f2e2917bcb888/geometry/proximity/make_sphere_mesh.h#L588-L591), it actively limits the amount of tessellation so that aggressively small resolution hints (which may not introduce overflow) will *still* not produce ridiculous meshes. (And, for the record, the [cylinder also limits refinement level](https://github.com/RobotLocomotion/drake/blob/b4070eacafdc3fbeab2cb83e080f2e2917bcb888/geometry/proximity/make_cylinder_mesh.h#L640-L650).)

The conclusion I draw is that we need to refactor the algebra a bit. Knowing the *maximum* refinement level, we need to compute the *minimum* allowable resolution hint. Then we always use `resolution_hint = max(allowable_resolution_hint, resolution_hint_in)`.

SeanCurtis-TRI, 2022-06-03:
It would also be good if they could get feedback during parsing indicating that the value they've provided is too small, and an alternate value has been used.

SeanCurtis-TRI, 2022-06-03:
@joemasterjohn Quick question. I haven't played with the reproduction example. But, presumably, the problem here is that when the quantity overflows, the user ends up with the *minimally* tessellated cylinder, right? As near as I can tell, when I call `static_cast<int>(value)` for `value > INT_MAX`, it silently resolves to -2147483648. Is that was inferred from the original problem? Is that the peg was being represented by a triangular prism?

jwnimmer-tri, 2022-06-06:
Because [Drake is a library](https://github.com/RobotLocomotion/drake/issues/17081#issuecomment-1129449897), I don't think a "warning" is appropriate here.

If Drake deems this input to be valid (by clamping it and _specifying_ that we will clamp it), then nothing has gone wrong and there is no need to bother the user at all.

If Drake deems this input to be invalid (by throwing an exception or emitting a structured parser error), then that error message can explain the details of the invalid input.

Either way, we don't need a "warning" while constructing the model.

In our parser operations, we reserve "warning" for cases where the user's input is correct and consistent with the _input specification_ but where that specification is not yet correctly implemented by Drake.  For example `mechanicalReduction` is a floating point value in URDF, but Drake's MbT does not model this concept yet.  If the user provides a valid, non-unity value in their input file (e.g., `2.0`), then Drake will _warn_ that such a feature is un-implemented.  Users parsing in "strict" model will receive an exception, but users by default would only receive a log message.  This reduces the boot-up effort required when users bring over a URDF from another ecosystem (e.g., ROS) that is working for them already -- if Drake is lacking some features, they can still at least try to do kinematics or similar, without needing to rewrite their model files immediately.

In case the input schema is a _Drake_ addition (as with the resolution hint in this case), then that concern does not apply.  We are the first-party implementation in this case.  When a Drake-specific input is invalid, we can just error out directly.

jwnimmer-tri, 2022-06-06:
I also meant to add: if we want to help the user with debugging their mesh resolution hint, the story for that should not be console spam, it should be _visualizing_ the collision meshes.

jwnimmer-tri, 2022-06-06:
Empirically, this also seems like something that users might hit frequently (as evidenced by someone actually hitting it!), and takes a _long_ time for them to debug it.

Could I ask dynamics to consider promoting it to "priority: high" and see if you can fix the overflow relatively quickly?  It seems like it would be only a few lines of code to check and clamp, plus a small unit test.

SeanCurtis-TRI, 2022-06-06:
As I think about what it means to "fix" this, I have several thoughts:

1. Generally, we have the question "how easy is it to observe your contact geometry"? (As @jwnimmer-tri suggested.) Making this as easy as possible would go a long way for users to understand what they're working with. Inferring geometry from contact surfaces is notoriously fraught.
2. In this case, the bug is that the user has requested a ridiculously high resolution model (as indicated by a ridiculously small resolution hint). So, this should be treated exactly the same as a non-overflowing resolution hint that would nevertheless trigger a mesh that is higher than we deem appropriate (see the code where we explicitly limit the refinement levels). So, both cases should cause the mesh to saturate at the maximum resolution. Performance would suck, but at least the contact surface would match what they expect to see.

DamrongGuoy, 2022-06-06:
@joemasterjohn Thank you for working on this issue.