RobotLocomotion/drake #17246: [examples/quadrotor] QuadrotorPlant dynamics are incorrect.

RussTedrake, 2022-05-22:
The unit test in `quadrotor_dynamics_test.cc` has the line:
```
  EXPECT_TRUE(ge_pose.rotation().IsNearlyEqualTo(ge_pose.rotation(), 1e-10));
```
Unsurprisingly, this passes.  But if you change it to the intended
```
  EXPECT_TRUE(ge_pose.rotation().IsNearlyEqualTo(mb_pose.rotation(), 1e-10));
```
Then the test fails.

I found this because I was writing up a (not so) quick example of the 3D dynamics of a quadrotor using Propeller.  Which gives different answers.
```
def MakeMultibodyQuadrotor():
    builder = DiagramBuilder()
    # The MultibodyPlant handles f=ma, but doesn't know about propellers.
    plant = builder.AddSystem(MultibodyPlant(0.0))
    model_instance = Parser(plant).AddModelFromFile(
        FindResourceOrThrow("drake/examples/quadrotor/quadrotor.urdf"))
    # By default the multibody has a quaternion floating base.  To match
    # QuadrotorPlant, we can manually add a FloatingRollPitchYaw joint.
    AddFloatingRpyJoint(plant, plant.GetFrameByName("base_link"),
                        model_instance, use_ball_rpy=False)
    plant.Finalize()

    # Now we can add in propellers as an external force on the MultibodyPlant.
    body_index = plant.GetBodyByName("base_link").index()
    # Default parameters from quadrotor_plant.cc:
    L = 0.15     # Length of the arms (m).
    kF = 1.0     # Force input constant.
    kM = 0 #0.0245  # Moment input constant.

    # Note: Rotors 0 and 2 rotate one way and rotors 1 and 3 rotate the other.
    prop_info = [
        PropellerInfo(body_index, RigidTransform([L, 0, 0]), kF, kM),
        PropellerInfo(body_index, RigidTransform([0, L, 0]), kF, kM),
        PropellerInfo(body_index, RigidTransform([-L, 0, 0]), kF, kM),
        PropellerInfo(body_index, RigidTransform([0, -L, 0]), kF, kM),
    ]
    propellers = builder.AddSystem(Propeller(prop_info))
    builder.Connect(propellers.get_output_port(),
                    plant.get_applied_spatial_force_input_port())
    builder.Connect(plant.get_body_poses_output_port(),
                    propellers.get_body_poses_input_port())
    builder.ExportInput(propellers.get_command_input_port(), "u")

    return builder.Build(), plant

# TODO(russt): Move this to Drake as a unit test.
def MultibodyQuadrotorExample():
    mbp_plant, mbp = MakeMultibodyQuadrotor()

    # Default parameters
    m = 0.775
    L = 0.15
    I = np.diag((0.0015, 0.0025, 0.0035))
    kF = 1.0
    kM = 0 #0.0245
    hand_derived_plant = QuadrotorPlant(m, L, I, kF, kM)

    # Compare the dynamics at a handful of states.
    mbp_plant_context = mbp_plant.CreateDefaultContext()
    mbp_context = mbp.GetMyContextFromRoot(mbp_plant_context)
    hand_derived_context = hand_derived_plant.CreateDefaultContext()

    MbpStateView = MakeNamedViewState(mbp, "mbp_state")

    rng = np.random.default_rng(seed=1037)
    for i in range(5):
        u = 0*rng.random((4,))
        x = rng.random((12,))
        # Note: The floating RPY joint in MultibodyPlant uses angular velocity,
        # not rpy derivatives, for the velocity variables.
        mbp_plant.get_input_port().FixValue(mbp_plant_context, u)
        mbp_context.SetContinuousState(x)
        mbp_xdot = mbp_plant.EvalTimeDerivatives(
            mbp_plant_context).CopyToVector()

        hand_derived_plant.get_input_port().FixValue(hand_derived_context, u)
        hand_derived_context.SetContinuousState(x)
        hand_derived_xdot = hand_derived_plant.EvalTimeDerivatives(
            hand_derived_context).CopyToVector()

        assert np.allclose(
            mbp_xdot, hand_derived_xdot
        ), f"\nmbp\t\t = {mbp_xdot}\nhand_derived\t = {hand_derived_xdot}"

MultibodyQuadrotorExample()
```
This also fails, but it passes if I set `x[9:] = 0`.  The dynamics don't match if I have non-zero angular velocity.

The multibody notation in [this line](https://github.com/RobotLocomotion/drake/blob/53571e1d65d1716d8b4b72ae8d3571ee89fa36cf/examples/quadrotor/quadrotor_plant.cc#L107) is wrong:
```
const Vector3<T> w_BN_B = rpy.CalcAngularVelocityInChildFromRpyDt(rpyDt);
```
According to that method's documentation, it should return `w_NB_B`, not `w_BN_B`.  Unfortunately, I'm timing out now.  Can I ask for the dynamics team to resolve it the existing failing test in `quadrotor_dynamics_test.cc`, please?

cc @sherm1 in case a different assignee is more appropriate.  
cc @mpetersen94 , in case it has any bearing on the strange results you've been seeing.

sherm1, 2022-05-22:
Paul @mitiguy can you figure out what's going on here?

mitiguy, 2022-05-23:
@sherm1   I am starting to look at it now.

mitiguy, 2022-05-26:
Closed by PR #17254

closed 2022-05-26 by mitiguy