RobotLocomotion/drake #17257: SDFormat parser doesn't warn for unsupported geometry shapes that are recognized by libsdformat

rpoyner-tri, 2022-05-24:
Symptom: input file geometry does not produce SceneGraph geometry, but neither does it give a warning to the user.

To reproduce: input file contains a geometry shape (in either visual or collision geometry) that is recognized as legal by libsdformat. Known example is `heightmap`, there are probably others.

For example, this patch:
```
diff --git a/manipulation/models/iiwa_description/iiwa7/iiwa7_no_collision.sdf b/manipulation/models/iiwa_description/iiwa7/iiwa7_no_collision.sdf
index 82e5d9e93a..58fef0bfaa 100644
--- a/manipulation/models/iiwa_description/iiwa7/iiwa7_no_collision.sdf
+++ b/manipulation/models/iiwa_description/iiwa7/iiwa7_no_collision.sdf
@@ -14,6 +14,11 @@
           <izz>0.0178</izz>
         </inertia>
       </inertial>
+      <collision name='badness'>
+        <geometry>
+          <heightmap/>
+        </geometry>
+      </collision>
       <visual name="iiwa_link_0_visual">
         <geometry>
           <mesh>
```
produces no heightmap geometry (because Drake does not support that yet), but also passes `//bindings/pydrake:py/parse_models_test` with no errors or warnings.

Victory condition: discover all of the geometry names with this symptom, and modify the parser to produce "geometry blah ignored"-type warnings.

rpoyner-tri, 2022-05-24:
\cc @sammy-tri feel free to take this, if you are working in adjacent SDFormat parser code.

EricCousineau-TRI, 2023-01-12:
Per SDFormat VC, plan is for @aaronchongth to tackle this.
@rpoyner-tri @jwnimmer-tri Please let us know if this redundant with y'all's efforts.

closed 2023-01-18 by jwnimmer-tri