RobotLocomotion/drake #17262: m1 Everything Builds Should Disable Unsupported Components

svenevs, 2022-05-25:
Relates: #16380, #17009.

Everything [build log here](https://drake-jenkins.csail.mit.edu/view/Mac%20M1/job/mac-m1-monterey-unprovisioned-clang-bazel-experimental-everything-release/6/console).

- [ ] Until drake is updated to mosek 10 (#17026), we should probably just disable `mosek` from `--config everything` on m1.  Snippets of interest:

    ```
    [12:05:42 AM]  Undefined symbols for architecture arm64:
    [12:05:42 AM]    "_MSK_appendbarvars", referenced from:
    [12:05:42 AM]        drake::solvers::MosekSolver::DoSolve(drake::solvers::MathematicalProgram const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, drake::solvers::SolverOptions const&, drake::solvers::MathematicalProgramResult*) const in lib_mosek_solver_compiled_cc_impl.a(mosek_solver.pic.o)
    [12:05:42 AM]    "_MSK_appendcone", referenced from:
    [12:05:42 AM]        drake::solvers::MosekSolver::DoSolve(drake::solvers::MathematicalProgram const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, drake::solvers::SolverOptions const&, drake::solvers::MathematicalProgramResult*) const in lib_mosek_solver_compiled_cc_impl.a(mosek_solver.pic.o)
    ```

    and

    ```
    [11:56:44 PM]  INFO: From Linking bindings/pydrake/common/test/text_logging_test.cpython-39-darwin.so:
    [11:56:44 PM]  ld: warning: ignoring file /Users/admin/workspace/mac-m1-monterey-unprovisioned-clang-bazel-experimental-everything-release/_bazel_admin/a3eee3f997801ec92a5f9ae18c4bb647/external/mosek/tools/platform/osx64x86/bin/libmosek64.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
    ```
- [ ] There is some complaining about `ipopt`, I'm not sure if this can be ignored or if we should do something.

    ```
    [12:03:32 AM]  ld: warning: object file (bazel-out/darwin_arm64-opt/bin/external/snopt/libfortran_objects.lo(snopt_wrapper.f90.pic.o)) was built for newer macOS version (12.4) than being linked (12.3)
    [12:03:32 AM]  ld: warning: object file (bazel-out/darwin_arm64-opt/bin/external/snopt/libfortran_objects.lo(np02lib.f.pic.o)) was built for newer macOS version (12.4) than being linked (12.3)
    [12:03:32 AM]  ld: warning: object file (bazel-out/darwin_arm64-opt/bin/external/snopt/libfortran_objects.lo(npopt.f.pic.o)) was built for newer macOS version (12.4) than being linked (12.3)
    ```

jwnimmer-tri, 2022-05-25:
> We should probably just disable mosek from --config everything on m1.

Agreed.

> ... complaining about ipopt ...

I seem to recall seeing that on other platforms / situations as well.  As long as the unit tests pass, I think we can ignore it.

svenevs, 2022-07-25:
Sorry for missing this.  We can close this, last licensing issue related to everything builds is #17565.

This is documented internally but FWIW mosek tests needed the ability to download from s3 as part of the orka image creation.  We download the mosek license on the fly, but the gurobi one is expected to live somewhere specific.  We may want to unify the approach in the future for simplicity (e.g., just download both of them).

closed 2022-07-25 by svenevs