RobotLocomotion/drake #17277: Remove redundant data default_mass_ in body.h  default_mass() from default_spatial_inertia_ in rigid_body.h

mitiguy, 2022-05-26:
There seems to be redundant data in body.h and rigid_body.h.
In body.h, the data field default_mass_ stores a body's default_mass.
There is an accessor method:
 double get_default_mass() const { return default_mass_; }

In rigid_body.h, the default mass is returned through a different storage mechanism.
  /// Returns the default value of this body's mass.  This value is initially
  /// supplied at construction when specifying this body's SpatialInertia.
  /// @returns This body's default mass.
  double default_mass() const {
    return default_spatial_inertia_.get_mass();
  }

sherm1, 2022-05-26:
Assigned to Alejandro for disposition.

mitiguy, 2022-05-27:
FYI: I am happy to push this forward in a PR, after hearing what Alejandro things.

sherm1, 2022-05-31:
Also aren't we just now adding DeformableBody? We need a Body base class if we're going to have both Rigid and Deformable.

amcastro-tri, 2022-06-02:
I am not sure it is the right abstraction though. 
For deformables we introduced `DeformableModel`, which is a `PhysicalModel`. That should do for things like FEM, MPM and even things like fluids in the future. 
That'd the leave all the multibody stuff into the realm of "rigid" only, which simplifies things a lot.

mitiguy, 2022-07-08:
Together, PR https://github.com/RobotLocomotion/drake/pull/17528 and PR https://github.com/RobotLocomotion/drake/pull/17472 resolve issue https://github.com/RobotLocomotion/drake/issues/17277.

closed 2022-07-12 by mitiguy