RobotLocomotion/drake #17280: Write MathematicalProgram to MathOptFormat in JSON

hongkai-dai, 2022-05-27:
[MathOptFormat](https://arxiv.org/pdf/2002.03447.pdf) is a unifying format to describe an optimization program in JSOn format, and it could be loaded by JuMP in Julia. We would like to support writing MathematicalProgram to this MathOptFormat. One benefit is that other languages (such as JuMP) can then load MathOptFormat and solve it with solvers not supported inside Drake (for example some people prefer using IBM Xpress, which Drake doesn't support).

cc @RussTedrake

RussTedrake, 2022-05-27:
Ironic that this came in almost perfectly synchronized with #17283.

hongkai-dai, 2022-05-27:
MathOptFormat uses variable name string to differentiate variables. So we will need unique name associated with each variable. This is discussed in https://github.com/RobotLocomotion/drake/issues/17068

jwnimmer-tri, 2022-05-28:
Even before we fix #17068, we can append the VariableId to the name to make distinct names for the any output format.

hongkai-dai, 2022-05-28:
> Even before we fix https://github.com/RobotLocomotion/drake/issues/17068, we can append the VariableId to the name to make distinct names for the any output format.

Good idea, I will just use `x` + VariableId for the moment in the json file.

jwnimmer-tri, 2023-05-22:
\CC @RussTedrake FYI as related to saving GCS programs to disk to try them with other solver packages.

AlexandreAmice, 2024-02-16:
I think this would be very good. A lot of the optimization community is converging on JuMP and a lot of new solvers are written/tested in Julia these days. It would be nice to be able to try these solvers on some Drake instances before doing the work to support them in Drake.