RobotLocomotion/drake #17369: benchmark tool should not assume intel_pstate driver is running

rpoyner-tri, 2022-06-09:
From slack thread here: https://drakedevelopers.slack.com/archives/C0KDGF4V9/p1654717537423909

Apparently there are developer Ubuntu machines on which the intel_pstate driver is missing or disabled. The benchmark tool should adapt to this. Possibly the variance of experiments will be worse in this situation. All of the typical performance measurement caveats apply (don't compare apples/oranges, etc.); it will still be up to the experimenter to analyze the statistics of their experiments.

Victory condition: benchmark experiments don't crash when intel_pstate is missing or disabled.

\cc @jwnimmer-tri @aykut-tri

jwnimmer-tri, 2022-06-09:
I'd like to suggest an amendment to the victory condition:

If we can't disable turbo boost, then there's no point in using the experiment runner in the first place.  The results will not be meaningful.  I think the victory condition should be "experiment runner disables turbo, or else fail-fast with a good message".  If we can adapt the code to disable turbo via another mechanism, great!  If not, we'll need to advise the user how to unbreak their computer, instead.

rpoyner-tri, 2022-06-09:
AFAICT, we have no way to read turbo states if the driver is missing. So the victory condition then becomes: "fail-fast` with a good message". Any other victory condition drags us down the rathole of detailed machine configuration.

closed 2022-06-09 by jwnimmer-tri

RussTedrake, 2023-03-05:
@jwnimmer-tri said the victory condition includes "If not, we'll need to advise the user how to unbreak their computer, instead."  I hit the fail fast, but am unable to discover anywhere what to do about it.

jwnimmer-tri, 2023-03-05:
@RussTedrake could you share your error message?

If it's this one ...

https://github.com/RobotLocomotion/drake/blob/42d86973c41725ad4ef1848547374fa21263009f/tools/performance/benchmark_tool.py#L124-L128

... could you share the machine type (AMD CPU, macOS, etc).

I guess we have some AMD CPUs in homecart workstations maybe?  It'd be nice to teach the tool how to control turbo on those.

RussTedrake, 2023-03-05:
Yes, that's the error message I see.

```
russt@home-cart13:~/drake$ uname -a
Linux home-cart13 5.15.0-60-generic #66-Ubuntu SMP Fri Jan 20 14:29:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```
and here is my [/proc/cpuinfo](https://github.com/RobotLocomotion/drake/files/10892369/cpuinfo.txt)

jwnimmer-tri, 2023-03-05:
=> https://github.com/RobotLocomotion/drake/issues/18942

PR #17371 (2022-06-09): [tools] benchmark_tool: Fail fast if we can't control turbo settings
Closes: #17369

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17371)