RobotLocomotion/drake #17375: ContactResults is not fully compliant as an AbstractValue

SeanCurtis-TRI, 2022-06-09:
### What happened?

The `ContactResult` class [contains a pointer to the `MultibodyPlant` that created it](https://github.com/RobotLocomotion/drake/blob/master/multibody/plant/contact_results.h#L57-L59). This is a value type that is provided on one of `MbP`'s output ports. When evaluating the output port (or a connected input port), we receive a `const ContactResult&`. As an `AbstractValue`, I should be able to *copy* that reference into a unique copy. That unique copy should now be *independent* of the system.

However, in the [copy constructor, the copied instance merely copies the pointer from the source](https://github.com/RobotLocomotion/drake/blob/master/multibody/plant/contact_results.cc#L50). This means, if the copy is persisted beyond the lifespan of the underlying `MbP` instance, the pointer will be defined, but invalid.

### Possible solutions
 - Null the `plant` pointer in the copy constructor; modify the documentation to indicate that the `MbP` pointer can be null for two reasons (default construct and copy).
     - It's worth emphasizing that while this preserves the raw data of `ContactResults`, it reduces the ability to *interpret* the data. The plant (and it's knowledge of the tree) is often necessary to fully assess the details of the results.
 - Define a set of queries on `ContactResults` data that would require the plant. Provide an API to support those operations.
     - The implementation of these APIs would directly invoke the `MbP` when the pointer is present.
     - The copy constructor would create a local copy of all the data necessary to support the queries. When the `MbP` pointer is null, it uses this cached data to support the set of queries.
     - Remove the method that returns the pointer to the parent `plant`.

### Version

aac420e3d24b8a57fa43b5f5f0fd212814454e5f

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

SeanCurtis-TRI, 2022-06-09:
### Possible solutions
 - Null the `plant` pointer in the copy constructor; modify the documentation to indicate that the `MbP` pointer can be null for two reasons (default construct and copy).
     - It's worth emphasizing that while this preserves the raw data of `ContactResults`, it reduces the ability to *interpret* the data. The plant (and it's knowledge of the tree) is often necessary to fully assess the details of the results.
 - Define a set of queries on `ContactResults` data that would require the plant. Provide an API to support those operations.
     - The implementation of these APIs would directly invoke the `MbP` when the pointer is present.
     - The copy constructor would create a local copy of all the data necessary to support the queries. When the `MbP` pointer is null, it uses this cached data to support the set of queries.
     - Remove the method that returns the pointer to the parent `plant`.

amcastro-tri, 2022-06-15:
I did a quick search. It'd seem that `ContactResults` does not use the plant pointer internally for anything, but it exposes it with `plant()`. It'd seem the only place where we use this is in `ContactVisualizer` to retrieve names? If that's the case, could it be that a good solution is that `ContactVisualizer` (or anyone consuming `ContactResults` really) keeps a reference to the plant that produces those results instead of the results themselves? That'd seem to allow getting rid of the plant pointer in `ContactResults`.

WDYT @SeanCurtis-TRI?

SeanCurtis-TRI, 2022-06-15:
Hmmm... A system defines an output port that contains a pointer to (const) itself. That seems as good as anything I can think of. It still means that the `ContactResults` contents are still largely obfuscated unless you have access to the right `MbP` instance to interpret.

How about another alternative. `ContactResults` is currently being used in two ways: as internal book-keeping for `MbP` to do its contact calculations. As such, its terse representation of the contact is sufficient for `MbP` because it serves as internal notes" that `MbP` is perfectly capable of processing.

However, it's also being used to communicate contact results outward to the world. Perhaps it would be better if we had introduced a new class `VerboseContactResults` that adds all of the missing information that external consumers of `ContactResults` want -- e.g., body names, model instance names, etc. The kinds of things that `ContactResultsToLcmSystem` is interested in. The fact that the verbose version is more expensive to compute is irrelevant because a) it wouldn't be computed at the same frequency as `ContactResults` is and b) that work is already being done by any *external* consumer of `ContactResults`.

jwnimmer-tri, 2022-06-15:
There's a need for the MbP back-pointer beyond just body names for visualization.  Some downstream systems want to compute kinematics, for example, or interrogate the tree structure.  Possibly we need a QueryObject-like interface for that later for efficiency (or correctness, in case the query depends on more contact data than just q and v), but in the meantime a simple back-pointer will do.

SeanCurtis-TRI, 2022-06-15:
I'd buy into the back pointer as the minimally sufficient solution to the problem. If we're agreed, I'll add it and deprecate the `ContactResults::plant()` API.

amcastro-tri, 2022-06-15:
ah... (face palm) thanks @jwnimmer-tri 

A pointer to the plant sounds good to me.

SeanCurtis-TRI, 2022-06-21:
A struct that contains the pointer would satisfy the requirement; but that gets exposed via the public API.

jwnimmer-tri, 2022-06-21:
For my prioritization -- is this on the critical path for anything?  (How important is it to complete this design discussion in a timely manner?)  This seems like a "type: cleanup" sort of problem, so doesn't seem particularly urgent.

SeanCurtis-TRI, 2022-06-21:
That's fine. I don't think it's blocking anything.

jwnimmer-tri, 2023-02-21:
> There's one constraint on a pointer-valued output port: [Value](https://github.com/RobotLocomotion/drake/blob/f58f78cb369c1838faa6a0c74753fffe1dcacb3f/common/value.h#L198-L200). We've explicitly disallowed pointers. A struct that contains the pointer would satisfy the requirement; but that gets exposed via the public API.

I think using a struct is better, even if Value did allow pointers.  The goings-on here are subtle and easy to make a mistake, so forcing the user to use longform syntax provides a good reminder that something special is happening, and the API documentation on the struct provides for a nice anchor to explain what's happening.

We could even imagine using a class instead of a struct, with nice exception messages when something goings wrong trying to access the underlying plant.