RobotLocomotion/drake #17385: [multibody] `model_directives.h` (only) uses namespace `multibody::parsing`

RussTedrake, 2022-06-12:
`model_directives.h` and `scoped_names.h` stand out as the only headers in `multibody/parsing` that use the namespace `multibody::parsing`.  All other headers/classes in this directory simply use the namespace `multibody`.  This causes small but unnecessary friction when having to remember/check the namespace for the various parsing methods.

Expected: these files should use namespace `multibody` only, like the other headers/classes in this directory.

cc @ggould-tri as the original author

jwnimmer-tri, 2022-06-12:
I would buy the main entry points `LoadModelDirectives` and `ProcessModelDirectives` moving into the simpler namespace, but I think many of the helper types in other files (e.g., `AddWeld` or `PrefixName`) are named too generically to live in `multibody` as-is.  For those names, the clarity that they are the `parsing` flavor of the name is important.  We could rename them as part of the move (so that they could stand on their own), but I'm not sure it's worth the effort.  I assume the major point of friction is only with the entry point?

RussTedrake, 2022-06-12:
Yes.  Moving only the entry points would resolve my concern.

rpoyner-tri, 2026-11-13:
Closing as not planned. As noted above in 2023, the better direction is to steer users toward the `Parser` API, which accepts `.dmd.yaml` model directive files directly, and to find out why people still use `process_model_directives.h`, rather than to move the namespace of a header we may want to deprecate. The `multibody::parsing` namespace there is a minor inconsistency and not worth the churn. If you hit a concrete problem porting to `Parser`, please open an issue about that.

closed 2026-11-13 by rpoyner-tri