RobotLocomotion/drake #17386: [multibody/parsing] ProcessModelDirectives uses std::abort which kills the jupyter kernel

RussTedrake, 2022-06-12:
Passing an invalid model directives yaml file from jupyter crashes the kernel.  For instance, I passed:

```
directives:
- add_model:
    name: homecart_bimanual
    file: package://anzu/models/homecart/homecart_bimanual.urdf
```
into ProcessModelDirectives, without having anzu in my package map, using e.g.
```
    parser = Parser(plant)
    directives = LoadModelDirectives(filename)
    added_models = ProcessModelDirectives(
        directives=directives, plant=plant, parser=parser)
```
and my kernel crashed without any consumable errors.  Reproducing in c++ revealed
```
[2022-06-12 10:27:42.353] [console] [error] Unable to resolve package 'anzu' for URI 'package://anzu/models/homecart/homecart_bimanual.urdf' using package map: 'PackageMap:
  - drake: /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/sandbox/darwin-sandbox/20/execroot/drake/bazel-out/darwin_arm64-opt/bin/manipulation/models/tri_homecart/parse_homecart_test.runfiles/drake
'
```
and looking at the model directives c++ code, I see uses of `std::abort`, which i believe are defeating our tools for trapping errors in the pydrake interface.

cc @ggould-tri as the original author.

jwnimmer-tri, 2022-07-19:
It seems like refactoring `ResolveModelDirectiveUri` to call `internal::ResolveUri` (from detail_path_utils) instead of re-implementing it by hand should be sufficient to resolve this.

Seems like a 5-minute fix, so tagging @rpoyner-tri or @SeanCurtis-TRI in case you have a few minutes downtime between other PRs.

SeanCurtis-TRI, 2022-07-19:
FTR I'm grabbing it.

PR #17604 (2022-07-20): [parsing] Handle unknown packages gracefully while processing model directives
The old behavior would attempt to resolve directive URIs by hand. This defers to the the utility function. Now, processing the directives no longer aborts, but throws an intelligible message.

closes #17386

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17604)
closed 2022-07-20 by EricCousineau-TRI