RobotLocomotion/drake #17390: Parser.AddAllModelsFromFile('my_model_directives.dmd.yaml')

RussTedrake, 2022-06-13:
It would be very nice to be able to parse model directives using the same API as we use for other file formats.

As discussed in https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1654939678154139, it might be nice to designate a more specific filename extension (e.g. `.dmd.yaml` instead of just `.yaml`).

calderpg-tri, 2022-06-27:
I would lean towards sticking with either of the common yaml extensions `.yaml` or `.yml` to get out-of-the-box editor highlighting for beginners. Multiple extensions like `.dmd.yaml` seems more likely to be confusing.

jwnimmer-tri, 2022-06-27:
I agree with `filename.endswith(".yaml")` as an invariant, so that editors work easily and out of the box.  I don't think using multiple extensions defeats that -- the editors I'm familiar with still treat it as YAML even with multiple extensions.

We _do_ need some kind of additional convention to clarify the schema of the file.

In some cases, projects can omit the extra extension stuff because other clues lead to the choice of schema.  For example, `docker-compose.yml` always has the same basename, so we know what its syntax is.  In other cases, there might be a farm of yaml files all under a given `conf` folder, which all share a uniform schema.

In our case, however, the model directives might be named anything, and appear in any folder, so we do need some kind of policyi for how to recognize them.

That could be with dual extensions like `/path/to/foo.model_directives.yaml` or with a required basename tail like `/path/to/foo_model_directives.yaml`.  I don't particularly care which naming policy we choose, but we do need something beyond `**/*.yaml`.

EricCousineau-TRI, 2022-06-28:
I think having multiple extensions, e.g. `*.dmd.yaml`, is completely fine, and I'd personally prefer that over basename suffix. Ideally, whatever rule we choose for the `Parser` routine, we make it required for directives in isolation.

Re; the full feature, my only concern for this is the nuance when a model directives file isn't encapsulated, and thus can't be parsed in isolation (or, more generally, in the wrong context).
Adding some text in a docstring or a tutorial would be helpful for when the questions inevitably come.

jwnimmer-tri, 2022-06-28:
@rpoyner-tri can you weigh in with your thoughts?  Then we can summarize / decide.

rpoyner-tri, 2022-07-14:
I'm fine with the `.dmd.yaml` proposal.

rpoyner-tri, 2022-07-21:
If we are moving model directives into multibody::Parser, do we need to maintain the legacy model directives interface at all? Can we deprecate and eventually delete it?

jwnimmer-tri, 2022-07-21:
It's not uncommon for code to have a `vector<ModelDirective>` already in memory (often created programmatically) that needs to be loaded.  We'll want to retain the ability to load that one way or another -- either via a Parser function that accepts that data type, or leaving `ProcessModelDirectives` as-is.  Possibly we could offer some sugar for going through `AddAllModelsFromString` instead (by serializing the directives vector into a string), but I suspect that would be awkward.

Other relevant piece is that some code sadly still relies on the `ModelInstanceInfo` details that (optionally) come out of directives processing.  For now, we need to preserve that information.  It would be plausible to have the `Parser` be able to regurgitate it, though, if we are switching up the API and want to get rid of the free functions.

rpoyner-tri, 2022-07-21:
My thought so far is to migrate the bulk of the model-directives implementation to detail/internal, and wire up the existing public api to that by trivial forwarding. That will solve my physical design problem, and make it easy to abandon the old api later if we decide it's time.

I'd appreciate thoughts on how to update README_model_directives.md, if we are keeping the old api around. Seems like the claims of "we'll ditch this all RSN" are no longer true. \cc @ggould-tri @EricCousineau-TRI

rpoyner-tri, 2022-07-21:
As for arbitrary compositing of models from multiple files, I wonder if the better answer is to invent some new class (possibly friends with parser, or something) that handles multi-file composites. The existing Parser is oriented around single-file ops (possibly multiple models, through), and I think that's a reasonable scope limit.

jwnimmer-tri, 2022-07-21:
Re: the README -- The claim of model directives being "temporary" is still true, but the time constant is not explained well.  Open Robotics is actively working on reaching feature parity (#15948, #17223) -- it's just not finished yet.

jwnimmer-tri, 2022-07-21:
With the addition of SDFormat includes (already implemented and available in Drake), the Parser already has the responsibility of loading multiple files (and models).  I think we should keep the Parser API for users the same.  If it makes sense under the hood to refactor into better helper classes, that's fine.

rpoyner-tri, 2022-07-21:
I'm not objecting to inclusions as part of Parser -- that ship has already sailed. The vector-of-files-with-external-joint-glue thing seems out of scope to me.

RussTedrake, 2022-08-29:
fwiw -- if this were to land this week, then it would clean up the notebooks that will give students their first look at Drake in class next week.  (not necessary, just know that there is interest!)

closed 2022-10-10 by rpoyner-tri