RobotLocomotion/drake #17398: Go To Name search improvements for website.

EvelynDixon-TRI, 2022-06-15:
API documentation can use improvements to the style and   results box for the Go To  search.

 the styling of hyperlinks from the "Go To Name" search box. They are "blue" as opposed to the new "red" color.  The alignment and color of the overlay box should be situated so that it doesn't cover the search string and has a better color match.  The image attached shows all 3 of these concerns: the hyperlink in the box with the search results is expected to be red,  the background & text colors between the results box and page should match better and finally the search string is covered. 

When scrolling to the end of the browser window while reviewing the C++ documentation, once you hit the footer area the search forms disappear.  This seemed to be problematic for someone who is perusing the documentation as you need to scroll all the way up (or hit the home key) to get the search back. The visual is jumpy. Jumpy Search form recording

There is a box that falls out of the window browser no matter the size.  Its content is not readable. This happens only in the search form "Go to name...". Go to name recording

EvelynDixon-TRI, 2022-06-15:
Attachments
[image]

https://user-images.githubusercontent.com/94084729/173714607-fb662ef6-8b0e-4fe6-85b4-2e094935f685.mp4

https://user-images.githubusercontent.com/94084729/173715113-50a092b1-73db-4f6a-ba01-2f0700a7813f.mp4

EvelynDixon-TRI, 2022-06-15:
Note that EchoCharlie has shared possible fixes with Jeremy for these issues.

jwnimmer-tri, 2022-06-17:
Here's the file I received: [Update 1.1.zip](https://github.com/RobotLocomotion/drake/files/8923545/Update.1.1.zip)

And the exposition I was given:

> Here's some updated files for doxygen, search.css, search.js and the extra css,
> - Search.css - simple color and space changes.
> - Search.js - positioning change of search window
> - Extra css - Added styling for search results, adjusted to overall style
>
> For the extra css file, all that's needs to be added there is this... at the end of that file.

```css
/* Search Results  */

.memtitle {
	background: none;
	border: 0;
}

.memproto, dl.reflist dt {
	background: #efefef;
	border: 1px solid #efefef;
	box-shadow: none;
	text-shadow: none;
}

.memdoc, dl.reflist dd {
	border: 1px solid #efefef;
	background-image: none;
	box-shadow: none;
}
```

closed 2024-11-01 by jwnimmer-tri