RobotLocomotion/drake #17409: Documentation unclear whether InverseDynamicsController provides gravity comp.

ggould-tri, 2022-06-16:
**Is your feature request related to a problem? Please describe.**
In Slack discussion with @JoseBarreiros-TRI he didn't know if `InverseDynamicsController` provided gravity compensation.
In looking for the answer, I had to read the implementation of `InverseDynamics` and `MultibodyTree` to prove it to my satisfaction.
Our documentation could be clearer on this point.

**Describe the solution you'd like**
Document that both `InverseDynamicsController` and `InverseDynamics` in full non-gravity-compensation mode provide gravity compensation.

**Additional context**
[Instigating conversation link](https://drakedevelopers.slack.com/archives/C2CK4CWE7/p1655406052111519)

jwnimmer-tri, 2024-12-31:
@RussTedrake I'm hopeful that #22328 made some progress here, but it's not clear to me whether there is more to do here.  Any thoughts?

RussTedrake, 2025-01-01:
I actually think it's been very clear on this point since https://github.com/RobotLocomotion/drake/commit/54b0625e0152ed40cfaad90e577b2887ae56c460#diff-eea6ac44faa8883dea77cac974fd2861f4b2ae10a69c4e5bbbbc92d43994ca7e ?  I'd say let's close it.  @ggould-tri -- feel free to reopen if you disagree.

(in reviewing the doc, I did find a few things to improve; i'll PR that separately now)

closed 2025-01-01 by RussTedrake