RobotLocomotion/drake #17426: InverseDynamicsController does not allow for output_projection_matrix which results in a not desired behavior

JoseBarreiros-TRI, 2022-06-21:
I noticed that IDC has an incorrect actuation mapping when used in the [following plant](https://github.com/JoseBarreiros-TRI/drake/blob/02a0c4840451acd57785c45570fd7a35b0551f47/examples/ruxid/python_humanoid_torso_idc_box/models/humanoid_torso_v2_noball_noZeroBodies_spring_prismatic.sdf):

[image]

From looking at the code, I believe the output needs to be multiplied by plant.MakeActuationMatrix() by default or at least allow the user to add the actuation_matrix in the arguments of the IDC in a similar way that is done for the PID controller (output_projection_matrix). I think the former is a better option because it reduces the overhead for the user. 

For context, connecting the IDC straight to the plant results in a not desired behavior:
[image]

and multiplying the output of the IDC by the actuation_matrix results in the desired behavior:
[image]

ggould-tri, 2022-06-21:
Relates fairly closely to #16908 (this may be a more concrete proposal to address that issue).
Relates tangentially to #12270 (also an issue of the layout of q / v vectors)

hongkai-dai, 2022-07-27:
Assigned to @siyuanfeng-tri as the original author of InverseDynamicsController.