RobotLocomotion/drake #17429: Add the ability to add WeldJoint() to a body that is not a "base"

JoseBarreiros-TRI, 2022-06-21:
Welding a base body to the world_frame can be done by adding a WeldJoint (see figure below), but this doesn't work for any other body that is not the base of the tree (e.g. waist). A method that rejjiters the kinematic tree to make any body the base, so a WeldJoint can be added, will improve usability.  

Drake suggests using a constraint: `RuntimeError: This mobilizer is creating a closed loop since the outboard body already has an inboard mobilizer connected to it. If a physical loop is really needed, consider using a constraint instead.` ; however, a [PositionConstraint](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_position_constraint.html) is not enforced as a weld.

[image]

sherm1, 2022-06-21:
Related: #13040

jwnimmer-tri, 2022-06-21:
@sherm1 In what way is this not a duplicate of #13040?  I can't seem to discern the difference between the two postings.

sherm1, 2022-06-21:
I read this one as wanting to create a loop, while #13040 is asking just to make a tree with a different base body. However, I see I may have misread this one -- @JoseBarreiros-TRI do you want to have to system welded to World in two places (forming a loop) or do you just want a tree with the welded body somewhere in the middle (like #13040)?

JoseBarreiros-TRI, 2022-06-21:
I want to add only one weld to a body that is not a base, like in [13040](https://github.com/RobotLocomotion/drake/issues/13040)

sherm1, 2022-06-21:
OK, let's keep this issue open since it is current. I'll close #13040 and #9939 (they will remain available for reference).

xuchenhan-tri, 2022-07-21:
IIUC, the current workaround is the same as that proposed in #13040. Since there's no immediate plan to add this feature, I'll put this one in the backlog for now.

sherm1, 2025-04-10:
@JoseBarreiros-TRI do you by any chance have the input file that caused this trouble? Or maybe a similar one? I'm close to having this fixed and would like to be able to try the fix against the original issue.

PR #22955 (2025-05-01): [multibody] Add tests to show issues 9939 & 17429 are closed
PR #22949 (now allowing reversed weld, revolute, and prismatic mobilizers) fixed long-standing issue #17429. The even-older issue #9939 (bad error message, somewhat related) is also resolved (sometime earlier).

This PR adds unit tests to prove that those two issues are resolved, and closes them.

Closes #9939
Closes #17429

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22955)
closed 2025-05-01 by sherm1