RobotLocomotion/drake #17433: Jupyter tests fail on Jammy

mwoehlke-kitware, 2022-06-21:
### What happened?

Jupyter tests are failing on Jammy:
- `//bindings/pydrake/multibody:py/examples/jupyter_widgets_examples_test`
- `//bindings/pydrake/systems:py/jupyter_widgets_examples_test`
- `//tools/jupyter:py/example_test`

The root cause appears to be the same in all cases.

### Version

ade445d67e232077aa2d2c805022238f5682dbc6

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
FAILED: //tools/jupyter:py/example_test (Summary)
      /root/.cache/bazel/_bazel_root/9ecb0f4ce434715009a64c73d901046c/execroot/drake/bazel-out/k8-opt/testlogs/tools/jupyter/py/example_test/test.log
      /root/.cache/bazel/_bazel_root/9ecb0f4ce434715009a64c73d901046c/execroot/drake/bazel-out/k8-opt/testlogs/tools/jupyter/py/example_test/test_attempts/attempt_1.log
      /root/.cache/bazel/_bazel_root/9ecb0f4ce434715009a64c73d901046c/execroot/drake/bazel-out/k8-opt/testlogs/tools/jupyter/py/example_test/test_attempts/attempt_2.log
INFO: From Testing //tools/jupyter:py/example_test:
==================== Test output for //tools/jupyter:py/example_test:
[IPKernelApp] WARNING | debugpy_stream undefined, debugging will not be enabled
[IPKernelApp] WARNING | Widget Javascript not detected.  It may not be installed or enabled properly. Reconnecting the current kernel may help.
Running notebook as a test (non-interactive)
Traceback (most recent call last):
  File "/root/.cache/bazel/_bazel_root/9ecb0f4ce434715009a64c73d901046c/sandbox/processwrapper-sandbox/16215/execroot/drake/bazel-out/k8-opt/bin/tools/jupyter/py/example_test.runfiles/drake/tools/jupyter/example_jupyter_py_main.py", line 15, in <module>
    main()
  File "/root/.cache/bazel/_bazel_root/9ecb0f4ce434715009a64c73d901046c/sandbox/processwrapper-sandbox/16215/execroot/drake/bazel-out/k8-opt/bin/tools/jupyter/py/example_test.runfiles/drake/tools/jupyter/example_jupyter_py_main.py", line 11, in main
    _jupyter_bazel_notebook_main(notebook_path, sys.argv[1:])
  File "/root/.cache/bazel/_bazel_root/9ecb0f4ce434715009a64c73d901046c/sandbox/processwrapper-sandbox/16215/execroot/drake/bazel-out/k8-opt/bin/tools/jupyter/py/example_test.runfiles/drake/tools/jupyter/jupyter_bazel.py", line 61, in _jupyter_bazel_notebook_main
    ep.preprocess(nb, resources={'metadata': {'path': notebook_dir}})
  File "/usr/lib/python3/dist-packages/nbconvert/preprocessors/execute.py", line 84, in preprocess
    self.preprocess_cell(cell, resources, index)
  File "/usr/lib/python3/dist-packages/nbconvert/preprocessors/execute.py", line 105, in preprocess_cell
    cell = self.execute_cell(cell, index, store_history=True)
  File "/usr/lib/python3/dist-packages/nbclient/util.py", line 84, in wrapped
    return just_run(coro(*args, **kwargs))
  File "/usr/lib/python3/dist-packages/nbclient/util.py", line 62, in just_run
    return loop.run_until_complete(coro)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 838, in async_execute_cell
    exec_reply = await self.task_poll_for_reply
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 631, in _async_poll_for_reply
    await asyncio.wait_for(task_poll_output_msg, self.iopub_timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
    return fut.result()
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 662, in _async_poll_output_msg
    self.process_message(msg, cell, cell_index)
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 917, in process_message
    self.handle_comm_msg(cell.outputs, msg, cell_index)
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 1009, in handle_comm_msg
    comm_object = handler(msg)
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 1062, in on_comm_open_jupyter_widget
    state = data['state']
KeyError: 'state'
================================================================================
==================== Test output for //tools/jupyter:py/example_test:
[IPKernelApp] WARNING | debugpy_stream undefined, debugging will not be enabled
[IPKernelApp] WARNING | Widget Javascript not detected.  It may not be installed or enabled properly. Reconnecting the current kernel may help.
Running notebook as a test (non-interactive)
[... 46 lines cut ...]
  File "/usr/lib/python3/dist-packages/nbconvert/preprocessors/execute.py", line 105, in preprocess_cell
    cell = self.execute_cell(cell, index, store_history=True)
  File "/usr/lib/python3/dist-packages/nbclient/util.py", line 84, in wrapped
    return just_run(coro(*args, **kwargs))
  File "/usr/lib/python3/dist-packages/nbclient/util.py", line 62, in just_run
    return loop.run_until_complete(coro)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 838, in async_execute_cell
    exec_reply = await self.task_poll_for_reply
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 631, in _async_poll_for_reply
    await asyncio.wait_for(task_poll_output_msg, self.iopub_timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
    return fut.result()
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 662, in _async_poll_output_msg
    self.process_message(msg, cell, cell_index)
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 917, in process_message
    self.handle_comm_msg(cell.outputs, msg, cell_index)
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 1009, in handle_comm_msg
    comm_object = handler(msg)
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 1062, in on_comm_open_jupyter_widget
    state = data['state']
KeyError: 'state'
================================================================================
```

jwnimmer-tri, 2022-06-21:
\CC @EricCousineau-TRI in case you have any tips

jwnimmer-tri, 2022-06-21:
https://github.com/jupyter/nbconvert/issues/1731 mentions the same error message, but I don't see any widgets metadata in our notebooks.

closed 2022-06-28 by jwnimmer-tri

EricCousineau-TRI, 2022-11-15:
To close loop back here, we don't encounter this in Anzu because we have `ipywidgets` provisioned.

> https://github.com/jupyter/nbconvert/issues/1731 mentions the same error message, but I don't see any widgets metadata in our notebooks.

I was able to see similar error when widgets are used, even if the metadata isn't present in version-controlled version of the notebook.

PR #17451 (2022-06-28): [jupyter] Customize ExecutePreprocessor to fix ipywidget tests
Something odd is happening with ipywidgets on Ubuntu 22.
I have no idea what's going on, but this gets us passing CI.

Closes #17433.

This curtails the widget message processing just prior to hitting this line in the #17433 backtrace.

```
  File "/usr/lib/python3/dist-packages/nbclient/client.py", line 917, in process_message
    self.handle_comm_msg(cell.outputs, msg, cell_index)
```

https://github.com/jupyter/nbclient/blob/12c7f0492beb9306192b609fbdab67bbac2e2b4b/nbclient/client.py#L459-L461

The commits here will be squashed, [... cut]