RobotLocomotion/drake #17440: CI: Investigate Mac cache server errors

BetsyMcPhail, 2022-06-22:
Reported in Slack thread [here](https://drakedevelopers.slack.com/archives/C270MN28G/p1655844149976379)

BetsyMcPhail, 2022-07-20:
**History**

Note: Using [mac-monterey-clang-bazel-continous-release](https://drake-jenkins.csail.mit.edu/view/Mac%20Monterey/job/mac-monterey-clang-bazel-continuous-release) as an example.

Sometime prior to April 21 ([#522](https://drake-jenkins.csail.mit.edu/view/Mac%20Monterey/job/mac-monterey-clang-bazel-continuous-release/522)) (that is the furthest Jenkins logs go back) the Mac cache server (`drake-mac-webdav`) is powered off (how??). Logs have the error message:  `*** Disabling remote cache because could NOT contact remote cache server`

On May 3 ([#559](https://drake-jenkins.csail.mit.edu/view/Mac%20Monterey/job/mac-monterey-clang-bazel-continuous-release/559)), the server was restarted via the VSphere interface. [drake-ci PR #151](https://github.com/RobotLocomotion/drake-ci/pull/151) updated the `DASHBOARD_REMOTE_CACHE ` location.

Logs then have the `REMOTE_CACHE_KEY` populated, as expected, _but_ now have errors like:

```
[9:59:47 PM]  WWARNING: Remote Cache: 4 errors during bulk transfer:
[9:59:47 PM]  com.google.devtools.build.lib.remote.http.HttpException: 500 Internal Server Error
[9:59:47 PM]  <html>
[9:59:47 PM]  <head><title>500 Internal Server Error</title></head>
[9:59:47 PM]  <body>
[9:59:47 PM]  <center><h1>500 Internal Server Error</h1></center>
[9:59:47 PM]  <hr><center>nginx</center>
[9:59:47 PM]  </body>
[9:59:47 PM]  </html>
```

Sometime between July 1 ([#752](https://drake-jenkins.csail.mit.edu/view/Mac%20Monterey/job/mac-monterey-clang-bazel-continuous-release/752/)) and July 5 ([#753](https://drake-jenkins.csail.mit.edu/view/Mac%20Monterey/job/mac-monterey-clang-bazel-continuous-release/753/)), the cache server was powered off again and we see `*** Disabling remote cache because could NOT contact remote cache server`

cc/ @svenevs

BetsyMcPhail, 2022-07-20:
Cache server has been powered on again.

svenevs, 2022-07-25:
Closing this as complete, provisioned vSphere jobs continue to use the cache server ([example](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-big-sur-clang-bazel-nightly-release/533/consoleFull)).  We'll need to rely on build cops continuing to check this weekly and report back.  It seems restarting the cache server is the most reliable route to success in the event of failure.

closed 2022-07-25 by svenevs