RobotLocomotion/drake #17457: Constructor for LinearSystem fails in python binding test.

xuchenhan-tri, 2022-06-27:
### What happened?

The mac unprovisioned CI has a failed test in https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-big-sur-unprovisioned-clang-bazel-nightly-release/. It is potentially caused by the new numpy version (1.23).

Possibly related: #12633

See https://drakedevelopers.slack.com/archives/C270MN28G/p1656345396339269 for early discussions.

### Version

0b90deb0553a1a8cfa392371ef5c4c49ec024912

### What operating system are you using?

macOS 11 (Big Sur), macOS 12 (Monterey)

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
Traceback (most recent call last):
  File "/Users/bigsur/workspace/mac-big-sur-unprovisioned-clang-bazel-nightly-release/_bazel_bigsur/bec0ae6dc20cf0d6dbe6178cb16ef78f/sandbox/darwin-sandbox/7951/execroot/drake/bazel-out/darwin-opt/bin/bindings/pydrake/systems/py/primitives_test.runfiles/drake/bindings/pydrake/systems/test/primitives_test.py", line 234, in test_linear_system_zero_size
    self.assertIsNotNone(LinearSystem(A, B, C, D))
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. pydrake.systems.primitives.LinearSystem_[float](A: numpy.ndarray[numpy.float64[m, n], flags.f_contiguous], B: numpy.ndarray[numpy.float64[m, n], flags.f_contiguous], C: numpy.ndarray[numpy.float64[m, n], flags.f_contiguous], D: numpy.ndarray[numpy.float64[m, n], flags.f_contiguous], time_period: float = 0.0)

Invoked with: array([], shape=(0, 0), dtype=float64), array([], shape=(0, 2), dtype=float64), array([], shape=(2, 0), dtype=float64), array([[0., 0.],
       [0., 0.]])
```

jwnimmer-tri, 2022-06-27:
If we can't resolve this the next day or two, then we should add a temporary work-around in our [setup scripts](https://github.com/RobotLocomotion/drake/blob/af2d88740e682952cf13692394e3afd9240d4a3d/setup/mac/binary_distribution/Brewfile#L23) to pin numpy to a known-working version.

rpoyner-tri, 2022-06-28:
I've got repro on a TRI internal mac, having done the `brew update` dance.

rpoyner-tri, 2022-06-28:
These manual steps will downgrade and pin `numpy` to 1.22.4:

```
# Must download the .rb file to sidestep some checksum step.
$ wget https://raw.githubusercontent.com/Homebrew/homebrew-core/f0f9916c02df84e5285c4cb77cbcaa963b15899d/Formula/numpy.rb
$ brew unlink numpy
$ brew install ./numpy.rb
$ brew pin numpy
```

closed 2022-06-30 by jwnimmer-tri