RobotLocomotion/drake #17483: Add Ubuntu Python 3.10 wheels

jwnimmer-tri, 2022-06-30:
With support for Jammy coming soon (#16217), we should also work on adding Python 3.10 wheel builds.

We expect to still compile them in a Focal Docker, but against a locally-installed Python 3.10 (not from Ubuntu).

mwoehlke-kitware, 2022-07-22:
> We expect to still compile them in a Focal Docker, but against a locally-installed Python 3.10 (not from Ubuntu).

We do? This is news to me. How do we plan to obtain this Python? This may involve considerable changes to the wheel building infrastructure if we aren't able to install from a PPA with "expected" package names.

I was under the impression that the intent was to build wheels in Jammy...

jwnimmer-tri, 2022-07-22:
I was hoping to offer Python 3.10 support without also increasing the minimum glibc requirement from 2.31 to 2.35.  Looking more in detail now, I can see how that might be difficult.  Let me rethink and get back next week.

mwoehlke-kitware, 2022-07-22:
Conversely... https://github.com/pypa/auditwheel/issues/388.

If we can just add a PPA to get 3.10 on Focal, that should be relatively straight-forward. As to actually *testing* the wheel on Jammy... *manual* testing shouldn't be too hard, but automating that the wheel should be built on Focal but tested on Jammy may be interesting.

jwnimmer-tri, 2022-07-25:
I found https://launchpad.net/~deadsnakes/+archive/ubuntu/ppa for 3.10 on Focal, but I don't want to use that as part of the release process.  I worry about its stability, and I was not a fan of the history of the project's issue discussions.

Could you help me estimate a WAG effort scoping for using locally-built Python 3.10 in the Focal docker?

I know that for testing we talked about building in Focal docker and then testing in Jammy docker.  That doesn't sound too onerous, just a PLATFORM=ubuntu:22.04 override?

For building, it _seems_ like ...

https://github.com/RobotLocomotion/drake/blob/987bdbc0f937d53bfb8a53af9d0810e800b07f86/tools/wheel/image/provision-python.sh#L12-L13

... is approximately the only part that needs to change in the build?  Does that sound right, and what might it look like to switch that to download the Python sources and build it from source?

mwoehlke-kitware, 2022-07-27:
> Could you help me estimate a WAG effort scoping for using locally-built Python 3.10 in the Focal docker?

Even if building it isn't exciting, setting up a usable environment might be. Also, are we going to roll our own 3.8, or is it going to depend on the Python version whether we use packages or roll our own?

I suppose I'll WAG "the better part of a week"... maybe more. Perhaps I'd be pleasantly surprised, but my suspicion is that it won't be trivial.

> I know that for testing we talked about building in Focal docker and then testing in Jammy docker. That doesn't sound too onerous, just a PLATFORM=ubuntu:22.04 override?

The problem is that the builder needs to gain the notion that the build platform and test platform aren't the same. For that matter, if we build on Focal, are we not going to (also?) test on Focal? It would seem a bit odd to not test on the same platform used to build...

jwnimmer-tri, 2022-08-22:
I took an hour and did a quick prototype of building Python 3.10 from source in a 20.04 Docker.  It seems to me like it will be a reasonable approach.  Here's the branch:

https://github.com/jwnimmer-tri/drake/commits/wheel-3.10

It has code for both building Python 3.10 from source in the 20.04 Docker while building the wheel, and then running Ubuntu 22.04 Docker with stock Ubuntu Python 3.10 for testing.  I haven't been able to run the testing code yet, though, because ...

Something goes awry with `setuptools` while packaging up the wheel.  Possibly my Python 3.10 install is wrong or incomplete, or some paths are misconfigured.  Or maybe `setuptools` for Python 3.10 does something differently that we need to accommodate.

In any case, this architecture seems like a reasonable path forward, at least a first attempt, so I'll send this back to @mwoehlke-kitware to dig in and see if you can get everything working.

mwoehlke-kitware, 2022-08-24:
So, the problem is this little... er, "gem":
```python
    ext_modules = [
        setuptools.Extension(name='drake',
                             sources=[]),
    ]
```

We're already not using this on macOS. But just removing it results (on Ubuntu) in a non-platform-specific wheel, which is all sorts of broken. We're already employing *all* of [the usual solutions](https://stackoverflow.com/questions/45150304), so I'm not sure what to do about this.

BTW, at some point we probably want to deal with Python's [non-distinction between data and importable things](https://github.com/pypa/setuptools/issues/3340).

jwnimmer-tri, 2022-09-14:
This appears to be finished.  The only remaining task it to announce in the release notes #17853.

closed 2022-09-14 by jwnimmer-tri