RobotLocomotion/drake #17524: Pattern for using constexpr std::array

amcastro-tri, 2022-07-07:
From @rpoyner-tri in the review of #17400:

> I feel like some style suggestions against using constexpr std::array might be in order, precisely because it's too easy to make uninitialized data with it

@jwnimmer-tri provided a nice pattern so that we get a compiler error:

```diff
--- a/multibody/plant/multibody_plant_config_functions.cc
+++ b/multibody/plant/multibody_plant_config_functions.cc
@@ -42,14 +42,19 @@ constexpr const char* ContactModelToChars(ContactModel contact_model) {
   }
 }
 
-constexpr auto MakeContactModelPair(ContactModel value) {
-  return std::pair(value, ContactModelToChars(value));
-}
-
-constexpr std::array<std::pair<ContactModel, const char*>, 3> kContactModels{{
-  MakeContactModelPair(ContactModel::kPointContactOnly),
-  MakeContactModelPair(ContactModel::kHydroelasticsOnly),
-  MakeContactModelPair(ContactModel::kHydroelasticWithFallback),
+struct ContactModelItem {
+  constexpr ContactModelItem(ContactModel value_in)
+    : value(value_in),
+      name(ContactModelToChars(value_in)) {}
+
+  const ContactModel value;
+  const char* const name;
+};
+
+constexpr std::array<ContactModelItem, 3> kContactModels{{
+  {ContactModel::kPointContactOnly},
+  {ContactModel::kHydroelasticsOnly},
+  {ContactModel::kHydroelasticWithFallback},
 }};
 
 // Take an alias to limit verbosity, especially in the constexpr boilerplate.
```

rpoyner-tri, 2022-07-07:
What's the victory condition for this issue?

jwnimmer-tri, 2022-07-07:
Fail at compile time when a developer screws up a magic number inside `multibody_plant_config_functions.cc`.

PR #17529 (2022-07-12): [plant] Shore up enum-string conversion implementation
Closes #17524.  Follow-up from #17400.

+@amcastro-tri for feature review, please.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17529)