RobotLocomotion/drake #17525: URDF/SDF defaults to dissipation_timescale = 0

amcastro-tri, 2022-07-07:
@jwnimmer-tri pointed out in [this Slack conversation](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1657230165865699) that users might find it cumbersome if `dissipation_timescale` is made required in their URDF/SDF files.

The solution proposed in that conversation is:
 1. Parsers encode a reasonable (though in 99% of the cases not what you want) default of `dissipation_timescale=0`, zero dissipation. The parser warns the users about this decision.
 2. MultibodyPlant still throws if `dissipation_timescale` is not provided for a particular geometry. 
 
 That way, users making models by hand will still need to assemble all the right pieces, but existing models in the wild would still run (with warnings, and maybe janky).

jwnimmer-tri, 2022-07-07:
FYI here's an example of issuing a geometry property warning:

https://github.com/RobotLocomotion/drake/blob/53e982e79a27df8484dad5aaa97b3f121b2527f0/multibody/parsing/detail_sdf_geometry.cc#L551-L559

amcastro-tri, 2022-07-11:
A number of thoughts, not precisely related to each other:
 1. Russ mentioned that he'd actually prefer the default to be in MBP, not the parser. Along with that, he'd prefer the warning to be issued by MBP, not the parser. That way the small 5% of users that set parameters programatically (I do that a lot with gflags) also get the benefit of a nice warning.
 2. Here is a failure mode. We set a default + emit a warning. Long log, TLDR, someone misses the warning. The users however wants to increase the amount of dissipation and does so by specifying the `drake::hunt_crossley_dissipation`. Nothing happens...
 3.  A user creates an MBP created with a config stating `discrete_contact_solver=tamsi`. The parser therefore does not even issue a warning when it sees the user specifies dissipation with `drake::hunt_crossley_dissipation`. The user is done parsing the model. Right after that, the user decides to call `plant.set_discrete_contact_solver("sap")`. 
 
 Thoughts?

rpoyner-tri, 2022-07-11:
Seems to me like this conversation is circling back to the idea of "do a bunch of cross-checks and issue warnings or errors (or explode) at Finalize() time", which feels like a recurring theme. I don't necessarily have a solution, just noting that the same kind of opportunities for error, and infelicities of trying to detect/report them keep arising.

rpoyner-tri, 2022-07-11:
FWIW, this is the current behavior of `atlas_run_dynamics` (no command line args) on master:

```
rico@puget-161804-10:~/checkout/drake$ bazel run //examples/atlas:atlas_run_dynamics 
INFO: Analyzed target //examples/atlas:atlas_run_dynamics (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //examples/atlas:atlas_run_dynamics up-to-date:
  bazel-bin/examples/atlas/atlas_run_dynamics
INFO: Elapsed time: 0.554s, Critical Path: 0.00s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
[2022-07-11 17:47:26.725] [console] [warning] /home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/execroot/drake/bazel-out/k8-opt/bin/examples/atlas/atlas_run_dynamics.runfiles/drake/examples/atlas/urdf/atlas_convex_hull.urdf:1362: warning: Skipping transmission since it's attached to a fixed joint "hokuyo_joint".
terminate called after throwing an instance of 'std::runtime_error'
  what():  No `dissipation_timescale` specified. For geometry atlas::Mesh1 on body l_foot. You are using a linear model of compliant contact that requires you to specify dissipation explicitly.
Aborted (core dumped)
```

I haven't tried, but i suspect modifying the model to get it to run would require spamming `dissipation_timescale` to many places.

amcastro-tri, 2022-07-11:
yes, it just bit me also. Definitely we want a default I believe. 
Where/how to issue a warning might happen at a later date?

What would you guys think if this "global" default was a `MultibodyPlantConfig` parameter? would that ease the documentation of this default at least?

jwnimmer-tri, 2022-07-11:
> Russ mentioned that he'd actually prefer the default to be in MBP, not the parser. 

But _why_?

Your premise is that the default value _is the wrong math_, in other words that there is _no such thing_ as a valid default -- any default is _absurd on its face_.

If the default value was the _correct_ math, then we should put the default into MbP and _never warn about it_.  But since the default is _absurd_, the most we should do is attempt to load third-party SDFormat and URDF files with a compatibility shim (taking the absurd default as the best guess).  When a user is _writing Drake API calls_ we should not to absurd things by default.

> That way the small 5% of users that set parameters programmatically ... also get the benefit of a nice warning.

To be clear, in the case where we put the default in the Parser, the MbP would **not** continue along silently.  It will throw a detailed _exception_ message instead of a _warning_.  I don't understand why that's a problem.  If you are adding geometries by hand using API calls, you should expect that you need to do it carefully.  If the API is too verbose, we should make sugar.  In no case should a function call expect that a `drake::log()` makes it to user eyeballs.

> (3) A user creates an MBP created with a config stating discrete_contact_solver=tamsi ...

It's fine to think about the corner cases, but I think getting the 80% solution in place quickly is important.

I think it's OK not to warn in this case.  They'll get an exception and figure it out from there.  If they are using the Config struct pattern, the solver will always be set before we start adding geometries anyway.

> I do that a lot with gflags ...

Moving forward, don't use gflags.  Use only yaml (the Config helper structs).  I'll be rewriting all of our examples over the next few months to do that.  (You can still pass yaml settings one-off on the command line, it doesn't need to be in a file.)  The Config structs can properly encapsulate defaults and warnings in a way that bare gflags cannot.

I will jump on the hangout to clarity.

jwnimmer-tri, 2022-07-11:
Update from f2f: we think we can add the default to MbP, and never warn or throw about anything.  That would be best!

rpoyner-tri, 2022-07-19:
Is this issue now done?

amcastro-tri, 2022-07-19:
Yes, resolved by #17554. Thanks!

closed 2022-07-19 by amcastro-tri