RobotLocomotion/drake #17541: suggestion to add an interface to communicate with the library scikit-fuzzy

dvogureckiy99, 2022-07-10:
I think many people know the well-known github project for control systems based on fuzzy logic scikit-fuzzy. It would be useful to create an interface for the exchange (interaction) of this library with scikit-fuzzy. I tried to implement the link between also well-known cssl and scikit-fuzzy and it's shown here Fuzzy cart pole control, but I didn't do it very nicely, so I think it's useful to do it.

rpoyner-tri, 2022-07-12:
I made a guess that this is best captured in `component: planning and control`.  @hongkai-dai feel free to reassign.

rpoyner-tri, 2022-07-12:
Assigning to @hongkai-dai for triage and further delegation.

hongkai-dai, 2022-07-27:
I am not familiar with fuzzy logic control and haven't used scikit-fuzzy yet. I am not sure what is the action item here. What is the functionality of this interface? What Drake feature do you need when you interact Drake with scikit-fuzzy?

dvogureckiy99, 2022-07-27:
[Typical Fuzzy Control System](https://pythonhosted.org/scikit-fuzzy/auto_examples/plot_control_system_advanced.html#example-plot-control-system-advanced-py)

Many fuzzy control systems are tasked to keep a certain variable close to a specific value. For instance, the temperature for an industrial chemical process might need to be kept relatively constant. In order to do this, the system usually knows two things:

The error, or deviation from the ideal value

The way the error is changing. This is the mathematical first derivative; we’ll call it delta

From these two values we can construct a system which will act appropriately.

There is [API reference](https://pythonhosted.org/scikit-fuzzy/api/api.html). We are primarily interested in the [module control](https://pythonhosted.org/scikit-fuzzy/api/api.html).

hongkai-dai, 2022-07-28:
This doesn't require us to implement a new controller/planner, but to connect to the scikit-fuzzy software. @jwnimmer-tri I think this is more of an adoption project for external users. May I delegate this issue to you?

jwnimmer-tri, 2022-07-28:
As best I can tell, nobody on the Drake development team has experience with `scikit-fuzzy`, so we are not going to be the best people to push this forward.

I think if someone who is familiar with `scikit-fuzzy` wanted to collaborate with Drake's development team to help integrate the two projects, we would be happy to to support that with guidance and advice.

We wouldn't do that coding inside the https://github.com/RobotLocomotion/drake project, though.  It should be a separate project.  Whoever is interested in working on `scikit-fuzzy` + `pydrake` should open their own new GitHub repository with the integration code.  We could discuss the project in _that_ repository, and then also link to that repository from Drake's documentation so other people can discover it.  Possibly some Q&A related to this could also happen on StackOverflow, even without a dedicated GitHub project.

In terms of `pydrake` itself, in any case there is nothing to do here.  I'll close this as "not planned".

closed 2022-07-28 by jwnimmer-tri

dvogureckiy99, 2022-07-28:
Thank you, I understand you. Then, if there is more than one person who really needs it, it will be really useful to create a new repository for this project.