RobotLocomotion/drake #17551: Improve drake-packages s3 UX

svenevs, 2022-07-12:
> Russ brought up the good point that perhaps we can improve the UX in case someone is trying to access an older nightly build that we're planning on yanking.
> It does seem like this might be possible, per these instructions:
https://docs.aws.amazon.com/AmazonS3/latest/userguide/CustomErrorDocSupport.html
>
> I think we should try customizing 403.html for the packages bucket, to point to the https://drake.mit.edu/installation.html page (and possibly even mention that nightly builds expire after 45 days?).
>
> https://github.com/RobotLocomotion/drake/pull/17413#pullrequestreview-1018560562

Splitting into its own issue to track CC @RussTedrake @jwnimmer-tri 

It is also possible to create a [static website using s3](https://docs.aws.amazon.com/AmazonS3/latest/userguide/WebsiteHosting.html).  The incentive to do this would be so that somebody can visit e.g., https://drake-packages.s3.amazonaws.com/ and have a simple browseable interface to visit `drake/nightly` for example and see a listing of all the files in there.  Everything is publicly readable, but users without access to s3 have no idea what the names of the files are.  Just something very simple like this: https://ftp.gnu.org/gnu/gcc/

Tasks:

- [ ] Create custom 403 response.
- [ ] Possibly create browser access via a static website.  Investigation into billing would be needed.

rpoyner-tri, 2022-07-12:
Assigning to @jwnimmer-tri for further delegation.

svenevs, 2022-07-15:
AFAICT the [documentation](https://docs.aws.amazon.com/AmazonS3/latest/userguide/CustomErrorDocSupport.html) is somewhat misleading, that's nested under the "static website" part and you can only set a single "error document".  URLs of interest:

- [drake-packages.csail.mit.edu](https://drake-packages.csail.mit.edu/)
    - https required, this is configured via cloudfront
- [drake-packages.s3.amazonaws.com](https://drake-packages.s3.amazonaws.com/)
    - I modified the permissions to allow an XML listing of the contents
- [drake-packages.s3-website-us-east-1.amazonaws.com](http://drake-packages.s3-website-us-east-1.amazonaws.com/)
    - http only, this is the so called "static website"
    - this is the only one that the error document loads for currently, I cannot figure out how to make that so for the other urls.

A vain attempt at listing contents via the static website was made to no avail.  From the docs on https://github.com/rufuspollock/s3-bucket-listing it seems that the XML listing sites is how it will always operate.  I gave https://github.com/qoomon/aws-s3-bucket-browser a quick attempt too without any success.

However, noting that the bucket is (and always has been) public read, the current permissions on the bucket now make it so that a slightly more helpful message for the XML pages will be produced:

- https://drake-packages.csail.mit.edu/drake/nightly/drake-20170814-mac.tar.gz

    ```xml
    <Error>
    <Code>InvalidObjectState</Code>
    <Message>
    The operation is not valid for the object's storage class
    </Message>
    <StorageClass>GLACIER</StorageClass>
    <RequestId>D1F1QR10TE91T87K</RequestId>
    <HostId>
    Pe3sb9YTfNMYE1JtWi1DT65C4tOviz/2SCn6eV8eUtviwnTVhqfn9c3GZCkDNEQP/taMe0PBwqw=
    </HostId>
    </Error>
    ```

- https://drake-packages.csail.mit.edu/not_a_file.txt

    ```xml
    <Error>
    <Code>NoSuchKey</Code>
    <Message>The specified key does not exist.</Message>
    <Key>not_a_file.txt</Key>
    <RequestId>R6TP5DQ6YXWA92V2</RequestId>
    <HostId>
    0lhgl57WIgKSj6c2z41JsFkwqpQUZ7lKR4RHXzIxh/f9+pKBPUdtt05+IGxup7GyiQXOqWx9RqI=
    </HostId>
    </Error>
    ```

There is currently an `index.html` and `error.html` in the `drake-packages` s3 bucket, the following URLs get you the landing page that will redirect you back to the drake main website:

- https://drake-packages.csail.mit.edu/
- http://drake-packages.s3-website-us-east-1.amazonaws.com/

If somebody chooses to try and operate through the static website, they can achieve the error document with

- http://drake-packages.s3-website-us-east-1.amazonaws.com/not_a_file.txt

but realistically we don't advertise this static website and it is borderline useless (https://drake-packages.csail.mit.edu/not_a_file.txt will not load this error page).

We can continue investigating, but I don't think this is a good use of time -- I do not believe it is possible to get the custom error document for the mirror websites such as drake-packages.csail.mit.edu.  If we are happy with the current `index.html` and `error.html` then I suppose it would make sense to track them in git somewhere, maybe in `drake/doc/_drake_packages_s3`?

jwnimmer-tri, 2022-07-18:
> I suppose it would make sense to track them in git somewhere, ...

I don't see any need for that.  S3 is just as durable as git, and we don't need to do ongoing code inspection.  I would actually just nuke the fancy pages and make them vanilla htmls file with a simple link back to the main site (drake.mit.edu).  All of the logo and hover-menu stuff is just going to bitrot.  We don't want to think about ongoing maintenance there.

Once the index.html and error.html pages are simplified, we can close this issue as "won't fix".

svenevs, 2022-07-19:
Simplified landing pages with redirects visible here:

- index: https://drake-packages.csail.mit.edu or http://drake-packages.s3-website-us-east-1.amazonaws.com
- error: http://drake-packages.s3-website-us-east-1.amazonaws.com/not_a_file.txt

closed 2022-07-19 by svenevs