RobotLocomotion/drake #17560: Diagnose / fix clang CI builds getting killed

svenevs, 2022-07-13:
```
[5:57:37 PM]  clang: error: unable to execute command: Killed
[5:57:37 PM]  clang: error: linker command failed due to signal (use -v to see invocation)
```

Additional information: https://drakedevelopers.slack.com/archives/C270MN28G/p1654814303386219

jwnimmer-tri, 2022-07-14:
Here's some keyword search spam (I had trouble finding this issue today)...

ASan, Address Sanitizer, Memory, OOM, `--local_ram_resources`, `math(EXPR DASHBOARD_JOBS "7 * ${DASHBOARD_PROCESSOR_COUNT} / 8")`.

BetsyMcPhail, 2022-07-18:
| Date | Failing Job |
| ------------- | ------------- |
| May 31  | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/531  |
| May 31  | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/539 |
|  June 9  | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/595 | 
| June 14 | https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-clang-bazel-nightly-address-sanitizer/124 |
| June 20 | https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-clang-bazel-nightly-address-sanitizer/131 |
| June 27 | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/660 |
| June 29 | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/675 |
| Jul 7 |  https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/704 |
| Jul 8 | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/709 |
| July 11 | https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-clang-bazel-nightly-address-sanitizer/153 |
| Jul 12 | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/728 |
| Jul 12 | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/736 |
| Jul 13 | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/741 |
| Jul 14 | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/746 |
| Jul 14 | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/757 |
| Jul 15 | https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-clang-bazel-nightly-address-sanitizer/158 |
| July 17 | https://drake-jenkins.csail.mit.edu/view/Weekly%20Production/job/linux-focal-clang-bazel-weekly-mosek-address-sanitizer/23/ |

#17593 Merged

| Date | Failing Job |
| ------------- | ------------- |
| Jul 18 | https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/762 |

svenevs, 2022-07-19:
#17593 was reverted via #17602 and we have a new job count approach via https://github.com/RobotLocomotion/drake-ci/pull/174 that will start kicking off tonight / with the next round of continuous jobs.  :crossed_fingers:

BetsyMcPhail, 2022-07-26:
We haven't seen any more instances of this issue since the CI change was merged - is this issue ready to close?

jwnimmer-tri, 2022-07-26:
Fine by me.  We can easily reopen if need be.

closed 2022-07-26 by jwnimmer-tri